FreeCAD/FreeCAD #16905: Exception handling on macOS causes crashes

hyarion, 2024-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

## Description
There are crash issues on macOS when exceptions propagate through Qt's event handling mechanism. This problem is particularly evident in `Measurement.cpp` where exceptions are currently thrown and causes crashes.  While exceptions can be caught before it reaches qt code, it cannot be caught afterwards in `GUIApplication::notify`. The issue appears to be linked to Qt's handling of exceptions, as detailed in [QTBUG-97061](https://bugreports.qt.io/browse/QTBUG-97061). The Qt team has closed this as "Invalid" since the behavior is beyond their control and instead recommended ensuring all exceptions are handled before reaching Qt code.

[Thiago comments in QTBUG-97061](https://bugreports.qt.io/browse/QTBUG-97061?focusedId=588632&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-588632) draws the following analogy:
> Closing as Invalid because we don't recognise this as a bug. It's not under our direct control.
>
> Why it works in other platforms? "A broken clock is correct twice a day" (unless it's a 24-hour analog clock, but those are rare) applies: just because the code is buggy doesn't mean it always fails.

It seams like FreeCAD followed an old suggestion which has since been removed.

[Another comment on QTBUG-97061](https://bugreports.qt.io/browse/QTBUG-97061?focusedId=587568&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-587568):
> That recommendation is stale. It's never worked properly and I've recently removed it too (https://codereview.qt-project.org/c/qt/qtbase/+/374112).
>
> Sorry. Don't do it. ALWAYS catch your exceptions before they go back to Qt code.

The recommendation was removed in Qt6.

## Steps to reproduce
(I'm just using `QuickMeasure` as an example, it can happen elsewhere too):
1. Add `throw new Base::RuntimeError("This will crash");` as the first line in `QuickMeasure::processSelection()`.
2. Build, run, and open FreeCAD
3. Create a part
4. Click on the new part in the Tree view
Crash with the output `libc++abi: terminating due to uncaught exception of type Base::RuntimeError*`

## Expected Behavior
All exceptions should be caught and handled before reaching Qt's event loop, preventing the application from crashing.

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-09-29:
Just as a clarification for anyone trying to reproduce this - the behavior is Qt - platform specific. On Linux this issue therefore cannot be reproduced, you need a Mac for that.
On Linux you can pass exceptions through Qt and catch them afterwards (although Qt discourages from doing so - broken clock note above) - on Mac you cannot, and it will crash.  I don't know about Windows.

The previous bugfix for #16820 for example triggers an exception instead of a null pointer dereference, however there is no exception handler in QuickMeasure, so the exception propagates up and through Qt code, eventually caught as.
`Unhandled unknown exception caught in GUIApplication::notify.`

This safety net does not hold on a mac, leading to a crash. Referring #16888 - since that's one more thing that needs to be fixed in QuickMeasure

hyarion, 2024-09-30:
@yorikvanhavre I think we need to reopen this and discuss how to fix it.
The PR that referred to this one only informs that it is a problem, not how to fix it, that's what this issue is about.

hyarion, 2024-10-04:
I have a draft here on how it could be done: https://github.com/hyarion/FreeCAD/commit/1f6649ad3e5a52f49ef6d9ea109c64ad0aee8cca

The solution is:
* a static `TryAndReport(` method is created that captures exceptions in the same way as `GuiApplication::notify(` does.
* `TryAndReport(` takes a function/lambda as argument and runs it inside the try-block.
* Refactor event slots to move their logic into a lambda which `TryAndReport(` is called with.

The following code:
```c++
void handleEvent() {
    doStuff();
    doOtherStuff();
}
```

would be refactored as:

```c++
void handleEvent() {
    TryAndReport([&]{
        doStuff();
        doOtherStuff();
    });
}
```

To remove all potential macOS crashes, we need to add this in _all_ places where an exception might pass by. This would be a really big change.

hyarion, 2024-10-04:
It might also be possible to create a macro that does this for us.

Something like:
```c++
#define NO_CRASH_CONNECT(sender, signal, receiver, slot) \
    QObject::connect(sender, signal, receiver, [=]() { TryAndReport([&]() { slot(); }); })
```

Then everywhere we use `QObject::connect`, we could use this `NO_CRASH_CONNECT` macro instead:
```
// Original connect call
connect(button, &QPushButton::clicked, this, &MainWindow::handleButtonClick);

// Refactored with exception handling
NO_CRASH_CONNECT(button, &QPushButton::clicked, this, &MainWindow::handleButtonClick);
```

_(code in this comment has not been tested)_

chennes, 2024-10-05:
I very much do not like adding new macros to the codebase -- I'd prefer an alternative solution.

hyarion, 2024-10-06:
@chennes maybe we could do the same thing similar, but with a static function:
```c++
    static void SafeConnect(Sender sender, Signal signal, Receiver receiver, Slot slot) {
        auto safeSlotWrapper = [slot, receiver](auto&&... args) {
            GUIApplication::TryAndReport([=]() {
                if constexpr (std::is_invocable_v<Slot, Receiver>) {
                    std::mem_fn(slot)(receiver);
                } else {
                    std::mem_fn(slot)(receiver, std::forward<decltype(args)>(args)...);
                }
            });
        };
        QObject::connect(sender, signal, receiver, safeSlotWrapper);
    }
```

So instead of:
```c++
// Original connect call
connect(button, &QPushButton::clicked, this, &MainWindow::handleButtonClick);
```

You would write:
```c++
// Refactored with exception handling
GUIApplication::SafeConnect(button, &QPushButton::clicked, this, &MainWindow::handleButtonClick);
```

(The static function doesn't need to live in `GUIApplication`, that's just an example.)

We would need to have this on a lot of places through, and doing this is getting pretty far outside my comfort zone - I'm lambdas, mem_fn, decltype... I'm not 100% how it works and I think I need some help with this.

yorikvanhavre, 2024-10-06:
@wwmayer any opinion here?

wwmayer, 2024-10-07:
Adding SafeConnect is only a way to avoid a crash on macOS and possibly get some feedback where an exception was raised. Whenever finding a slot that raises an exception it must be fixed to stop doing it.

But then you can also mark all slot functions with `noexcept` and run clang-tidy over the code base to find all occurrences of `bugprone-exception-eascape`

wwmayer, 2024-10-07:
To still implement a wrapper for a safe connection here is the class Safe with several overloaded connect() methods:

```
class BaseExport Safe
{
public:
template<typename Func, typename ... Args>
static void tryInvoke(Func func, Args&& ... args)
{
    try {
        if constexpr (std::is_invocable_v<Func>) {
            func();
        }
        else {
            func(std::forward<decltype(args)>(args)...);
        }
    }
    catch (...) {
        printf("Raised exception\n");
    }
}

//connect to a function pointer  (not a member)
template <typename Func1, typename Func2>
static inline typename std::enable_if<int(QtPrivate::FunctionPointer<Func2>::ArgumentCount) >= 0, QMetaObject::Connection>::type
        connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal, Func2 slot)
{
    auto safeSlotWrapper = [slot](auto&&... args) {
        tryInvoke(slot, std::forward<decltype(args)>(args)...);
    };

    return QObject::connect(sender, signal, sender, safeSlotWrapper, Qt::DirectConnection);
}

//connect to a function pointer  (not a member)
template <typename Func1, typename Func2>
static inline typename std::enable_if<int(QtPrivate::FunctionPointer<Func2>::ArgumentCount) >= 0 &&
                                      !QtPrivate::FunctionPointer<Func2>::IsPointerToMemberFunction, QMetaObject::Connection>::type
        connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal, const QObject *context, Func2 slot,
                Qt::ConnectionType type = Qt::AutoConnection)
{
    auto safeSlotWrapper = [slot](auto&&... args) {
        tryInvoke(slot, std::forward<decltype(args)>(args)...);
    };

    return QObject::connect(sender, signal, context, safeSlotWrapper, type);
}

//connect to a functor
template <typename Func1, typename Func2>
static inline typename std::enable_if<QtPrivate::FunctionPointer<Func2>::ArgumentCount == -1, QMetaObject::Connection>::type
        connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal, Func2 slot)
{
    auto safeSlotWrapper = [slot](auto&&... args) {
        tryInvoke(slot, std::forward<decltype(args)>(args)...);
    };

    return QObject::connect(sender, signal, sender, safeSlotWrapper);
}

//connect to a functor, with a "context" object defining in which event loop is going to be executed
template <typename Func1, typename Func2>
static inline typename std::enable_if<QtPrivate::FunctionPointer<Func2>::ArgumentCount == -1, QMetaObject::Connection>::type
        connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal, const QObject *context, Func2 slot,
                Qt::ConnectionType type = Qt::AutoConnection)
{
    auto safeSlotWrapper = [slot](auto&&... args) {
        tryInvoke(slot, std::forward<decltype(args)>(args)...);
    };

    return QObject::connect(sender, signal, context, safeSlotWrapper, type);
}
};
```

and some test code

```
class BaseExport TestObject : public QObject
{
    Q_OBJECT

public:
    void testSignal() {
        Q_EMIT emitSignal(5);
    }
    void testSignalTwo() {
        Q_EMIT emitSignalTwo(5, 1.0);
    }
    void testSlot() {
        throw std::string("Exception");
    }
Q_SIGNALS:
    void emitSignal(int);
    void emitSignalTwo(int, double);
};

static inline void testSlot(int)
{
    printf("Slot func\n");
}

static inline void testSlotNoArg()
{
    printf("Slot func\n");
}

static inline void testExc(int)
{
    throw std::string("testExc");
}

static inline void testExcNoArg()
{
    throw std::string("testExc");
}

static inline void testFunc()
{
    TestObject* obj = new TestObject();
  //Safe::connect(obj, &TestObject::emitSignal, obj, &TestObject::testSlot);
    Safe::connect(obj, &TestObject::emitSignal, &testSlot);
    Safe::connect(obj, &TestObject::emitSignal, &testSlotNoArg);
    Safe::connect(obj, &TestObject::emitSignal, &testExc);
    Safe::connect(obj, &TestObject::emitSignal, &testExcNoArg);
    Safe::connect(obj, &TestObject::emitSignal, obj, &testSlot, Qt::AutoConnection);
    Safe::connect(obj, &TestObject::emitSignal, obj, &testExc, Qt::AutoConnection);
    Safe::connect(obj, &TestObject::emitSignal, [=](int val) { printf("Slot func: %d\n", val); });
    Safe::connect(obj, &TestObject::emitSignal, [=]() { printf("Slot func:\n"); });
    Safe::connect(obj, &TestObject::emitSignal, [=]() { throw std::string("Slot func:\n"); });
    Safe::connect(obj, &TestObject::emitSignal, obj, [=](int val) { printf("Slot func: %d\n", val); });
    Safe::connect(obj, &TestObject::emitSignal, obj, [=]() { printf("Slot func: \n"); });
    Safe::connect(obj, &TestObject::emitSignal, obj, [=]() { throw std::string("Slot func: \n"); });
    Safe::connect(obj, &TestObject::emitSignalTwo, &testExcNoArg);
    obj->testSignal();
    obj->testSignalTwo();
}
```

However, the most important overloaded method that allows to
`Safe::connect(obj, &TestObject::emitSignal, obj, &TestObject::testSlot);`
is not implemented because writing a wrapper for it is a bit more tricky since the slot function must be a member function of the receiver class that are passed to QObject::connect(). So, writing the wrapper function like
```
template <typename Func1, typename Func2>
static inline QMetaObject::Connection connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal,
                                 const typename QtPrivate::FunctionPointer<Func2>::Object *receiver, Func2 slot,
                                 Qt::ConnectionType type = Qt::AutoConnection)
{
    auto safeSlotWrapper = [receiver, slot](auto&&... args) {
        tryInvoke(slot, std::forward<decltype(args)>(args)...);
    };

    return QObject::connect(sender, signal, receiver, safeSlotWrapper, type);
}
```
is not possible because `safeSlotWrapper` isn't a member function of the type `QtPrivate::FunctionPointer<Func2>::Object`.

hyarion, 2024-10-07:
I wonder though if this just adds unnecessary complexity. Maybe noexcept and wrapping it manually with the TryAndReport-like structure would be a good start.

wwmayer, 2024-10-07:
What do you mean with wrapping _manually_?

hyarion, 2024-10-07:
> What do you mean with wrapping _manually_?

I mean like suggested here: https://github.com/FreeCAD/FreeCAD/issues/16905#issuecomment-2393865954

Pros with version above:
* no "magic"
* easy for people to understand
* less code to maintain

Pros with your solution:
* it's easy to just swap out connect in most cases

But maybe the following you wrote is preferred:
> But then you can also mark all slot functions with noexcept and run clang-tidy over the code base to find all occurrences of bugprone-exception-escape

wwmayer, 2024-10-08:
> it's easy to just swap out connect in most cases

Sorry, but that's not the case. In my implementation only the connect() variants are handled to support functors, lambdas and pointers to global functions but not pointers to member functions because that's not easy to realize. But the latter is the vast majority of how we use connect().

Further points to consider are:
* My implementation is based on Qt5. With Qt6 or future Qt versions the code possibly has to be adjusted. Or in the worst case it's not even possible any more.
* The code may not compile for all supported compilers. MSVC or the compilers on MSYS/MinGW are known to cause problems. The code was tested with clang on Linux.

Here we talked a lot about Qt's signal/slot mechanism but what about event handlers? Do we also need to handle exceptions there or will they we handled by `GUIApplication::notify`?

chennes, 2024-10-16:
Is there ongoing work to resolve this issue? I'm inclined to drop the "blocker" status if not.

hyarion, 2024-10-16:
There's no ongoing work.

I think the conclusion here is that everything needs to be caught, or we'll get a crash on macOS.
I doubt we will go through all code in one go to fix this.

Maybe the best way forward is to fix one module at a time where we find a new crash (or find a developer-build exception warning on non-mac builds).

Dropping the blocker status is ok by me

maxwxyz, 2024-10-17:
Removed blocker tag.