FreeCAD/FreeCAD #16910: Assembly WB: datum planes distance joint doesn't work as expected

sheetmetalman, 2024-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The distance joint between two planes does not work, as shown in the video.

1. First, I insert the column shoe; and the ground joint in the center of assembly;
2. Second, I insert the tube;
3. Third, I do the distance joint between the bottom de the tube and the upper surface of the coluumn shoe;
4. Fourth, I try to do the distance joint between the datum plane of the tube and the datum of assembly.

[FREECAD FILES.zip](https://github.com/user-attachments/files/17179970/FREECAD.FILES.zip)

https://github.com/user-attachments/assets/96f3aa44-6d77-46b2-a1ab-6178959d8e3a

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Assembly4 0.50.12
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-29:
Can you make the title less generic ?

luzpaz, 2024-10-04:
@furgo16 can you reproduce ?

AWeighell, 2024-10-30:
This is my first post. Apologies if I have got it wrong.
I agree with Sheetmetalman.  There is a problem with the Distance Joint in the Assembly WB.  It is difficult to be specific because the results are often not repeatable.  For example, in the attached, I am attempting to align planes of 2 assembly objects by selecting their planes or a plane with a vertex with a distance joint.  Alignment results with aligning different planes.  for example the non-anchored part might flip by 90 deg.
 
[100A Roof Assy.zip](https://github.com/user-attachments/files/17576313/100A.Roof.Assy.zip)
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.67
  * Assembly4 0.50.15
  * Curves 0.6.50
  * dodo 1.0.1
  * fasteners 0.5.29
  * flamingo
  * frame 0.1.1
  * lattice2 1.0.0
  * sheetmetal 0.5.3

MiqCG, 2024-10-30:
Hi,
Maybe I am not understanding correctly the problem but I think that in the first post there aren't Datum planes for attach the joints. https://wiki.freecad.org/Datum

Opening the original post files, I see the two objects and two visible Origin planes (one of them belonging to the assembly container). I cannot see any Datum plane attached to the bodies.

If I hide the Origin planes and create a Datum plane in each body, I get a correct distance constraint between the bodies. See and test with the modified files (beware that one datum plane is smaller than the tube).

Tested in 1.0RC2 and 1.1.0dev-39092

[Datum_MiqCG.zip](https://github.com/user-attachments/files/17577278/Datum_MiqCG.zip)

Uploading Datum.mp4…

luzpaz, 2024-10-30:
@MiqCG please edit the above comment re-attach the screencast ? TIA

AWeighell, 2024-10-31:
@MiqCG, Many thanks.  I think I am starting to figure it out.  These are my thoughts.  I am still running behind the bus.  I would welcome any comments and corrections.
  
I would suspect that this question will be a trip hazard for somebody has previously used Inventor.  e.g.   Somebody like me who jumps in without reading the manual.  Inventor allows every combination of alignment between Part origin planes / faces and assembly origin planes / Extra planes added to either part or assembly.  e.g.  Part origin planes to Assembly origin planes, part origin planes to part origin planes, assembly origin planes to faces etc.  FreeCAD does not.   
For the Assembly WB Fixed Distance Joints. 
FC does NOT allow joints with Assembly origin planes and part origin planes (or datum planes)?
FC allows joints between Part origin planes.
The aligment behaviour is not always consistent - sometimes parts do not align as expected.  For example, part to part plane origin plane alignment sometimes depends on the order that the planes are selected.  I am struggling to come up with repeatable examples to show this.  I will carry on investigating.

PaddleStroke, 2024-10-31:
My current PR https://github.com/FreeCAD/FreeCAD/pull/16675

Is going to introduce core datums objects and it is going to fix issues about use of origins.
So without having read in details I think this issue will be fixed by said PR.

sheetmetalman, 2024-10-31:
@AWeighell and everyones.

Sorry! I got the names mixed up.

It's origin planes, not datum planes.

MiqCG, 2024-11-01:
Ok, thanks.
So maybe @luzpaz can close this issue as a duplicate of https://github.com/FreeCAD/FreeCAD/issues/15995 ?

As I read the thread of https://github.com/FreeCAD/FreeCAD/pull/16675 Pull Request, we will have to wait until the release of 1.0 as this feature is part of a complicated merge.

Meanwhile, @sheetmetalman , you can use the Datum planes workaround ;-)

AWeighell, 2024-11-01:
@MiqCG, 
Many thanks.  The datum planes and knowing which planes work and which don't in the Assembly WB is good for me.

luzpaz, 2024-11-01:
Closing as duplicate of https://github.com/FreeCAD/FreeCAD/issues/15995

closed 2024-11-01 by luzpaz