FreeCAD/FreeCAD #16912: [TD]Smart dimension tool does not susequent selections

WandererFan, 2024-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Use the smart dimension tool to create a dimension.  After this,  you can only select the dimension's parent view or geometry within that view and new dimensions are created for every mouse click.   

The only way I could find to reset the selection mechanism is to click on an editable template field.

### Full version info

```shell
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38844 +2 (Git)
Build type: debug
Branch: fixAutoCorrectMsgMissing16892
Hash: d5dea2f2b83e13a5de75d63d0889712f47b9d5dd
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21
```

### Subproject(s) affected?

TechDraw

### Anything else?

This is the file I was using when I discovered this:

[Z_Platte-roy.FCStd.not.zip](https://github.com/user-attachments/files/17180146/Z_Platte-roy.FCStd.not.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-03:
@PaddleStroke FYI

PaddleStroke, 2024-10-03:
I am not sure to understand the issue, perhaps I am missing something.

The tool is working in continuous mode. ie it is active until you dismiss it by right clicking or pressing escape. The same as sketcher tools.

Is this the behavior you describe or is it something else? If it is something else please provide a video because I cannot reproduce on my side.

WandererFan, 2024-10-05:
forum topic: https://forum.freecad.org/viewtopic.php?t=91145

PaddleStroke, 2025-03-10:
Perhaps this should be closed since it is actually [17200](https://github.com/FreeCAD/FreeCAD/issues/17200) no?

closed 2025-03-10 by maxwxyz