FreeCAD/FreeCAD #16913: Cannot create a simple eclipse in sketcher

amreo, 2024-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem is that I cannot create a simple eclipse centered in the origin point. It impacts the user experience because the user doesn't understand why a simple eclipse cannot be "created".

Step to reproduce:
1) open freecad
2) go to Part Design
3) Create a new sketch on XY plane
4) Click on "create eclipse by center" (Sketcher_CreateEclipseByCenter)
5) Click to the origin point.
6) Click on the Y axis
7) Click on the X axis

Expected result: a unconstrained* eclipse centered around the origin point
Actual result: a zero sized (no visible eclipse) and a bunch of errors:

### Full version info

```shell
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * ExplodedAssembly
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

Sketcher

### Anything else?

Log:
```
9:51:27  Invalid solution from DogLeg solver.
19:51:27  Updating geometry: Error build geometry(0): 
19:51:27  Invalid solution from DogLeg solver.
19:51:27  Updating geometry: Error build geometry(0): 
19:51:27  Invalid solution from DogLeg solver.
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.edge1
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.edge1
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.edge1
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.edge1
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.edge1
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.edge1
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.edge1
19:51:32  <Sketch> SketchObject.cpp(11416): cannot find subname ;g1.
19:51:41  TempoVis.restore: failed to restore detail ('SDVProperty', 'Unnamed', 'XY_Plane.Visibility'): 'NoneType' object has no attribute 'ViewObject'
19:51:41  TempoVis.restore: failed to restore detail ('SDVProperty', 'Unnamed', 'XY_Plane.LinkVisibility'): 'NoneType' object has no attribute 'ViewObject'
19:51:41  TempoVis.restore: failed to restore detail ('SDVProperty', 'Unnamed', 'Sketch.Visibility'): 'NoneType' object has no attribute 'ViewObject'
19:51:41  TempoVis.restore: failed to restore detail ('SDVProperty', 'Unnamed', 'Sketch.LinkVisibility'): 'NoneType' object has no attribute 'ViewObject'
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-29:
Do you happen to mean 'ellipse' ?

amreo, 2024-09-29:
> Do you happen to mean 'ellipse' ?

Yes, I mean ellipse

> Edit: if so, then is probably a dupe of #16321

Yes, I think it is, too

luzpaz, 2024-09-30:
Closing as duplicate

closed 2024-09-30 by luzpaz