FreeCAD/FreeCAD #16914: External geometry cannot be offset

RoadsideMaker, 2024-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem - Sketcher Offset tool shows error on use when trying to create the offset from projected external geometry.
Error message - Selection has no valid geometries. B-splines and points are not supported yet.
Impact - Does not allow for fluid workflow when sketching clearances or other offset geometry.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Latvian/Latvia (lv_LV)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Design456 0.0.1
  * fasteners.backup1715429646.4893107 0.5.17
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-29:
Duplicate of #12461

mwganson, 2024-09-29:
As a workaround, the MeshRemodel SketchPlus object has the feature to show the External Geometry as normal geometry, which is applied after the sketch is closed.  It also has the offset feature.

Steps:

1) install MeshRemodel workbench in addon manager
2) Create SketchPlus object (if it is created with an existed sketch selected, it offers the option to copy the sketch contents, including constraints, attachment options, and links to external geometry).
3) Add the links to external geometry to the SketchPlus object
4) Close the sketch editor, right click the SketchPlus object in the tree, in the context menu select the top menu item, setup array or function -> show -> check external checkbox.
5) In the context menu, setup array or function -> offsetting -> setup offset in dialog.

Note, the additional features of the SketchPlus object are all applied after the sketch editor is closed, none of the features will appear in the sketch editor, they only appear after closing the sketch.

closed 2024-09-30 by Roy-043