FreeCAD/FreeCAD #16920: Part Design WB: Deselecting a Sketch Segment in Part Design Workbench Clears All Previous Selections (Issue Not Present in Version 0.21.2)

felixscheibe, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Part Design workbench, when selecting multiple segments of a sketch contour, deselecting a single segment causes all previous selections to be cleared. This is problematic, especially when many segments have already been selected, as the user has to start over. The expected toggle functionality (selection/deselection) does not work, as deselecting one segment immediately clears all previous selections.

This issue does not occur in version 0.21.2, where the functionality works as expected.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * Manipulator 1.5.7
  * sheetmetal 0.4.25
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-30:
Do you mind providing a screencast of this occurring?

depthoffocus, 2024-09-30:
This appears to be broken as far back as 38827, which is the oldest weekly I have here.

mwganson, 2024-09-30:
Not limited to Part Design or to Sketcher.  For example, make a default cube primitive in Part workbench.  Select some edges with Ctrl+Select, then try to deselect one of the selected edges with Ctrl+Select, result: all edges get deselected.

luzpaz, 2024-10-11:
@PaddleStroke can you check this out?

depthoffocus, 2024-10-19:
I have noticed the same thing occurs with the selection of faces, and when you experiment with faces you can much more easily see the behaviour.

It is difficult to explain in words. 

If you select three faces then unselect one, all the selected faces are cleared.  So you try again, but your next two face multiple-selections may not "take", depending on which ones you click. 

- Let's say you click to select faces, A, B, and C. 
- And then click to deselect A. So only B and C should be shown as selected
- But all the faces are presented as if they are deselected.
If you click face A, it will show reselected.
- If you click face B or C, the GUI will act as if they are selected, but they will be shown deselected after the click.

Without understanding the code, my impression is that the problem is that the state passed to the function that decides which faces to show as selected is out of sync with the state retained by the GUI for those that are _actually_ selected.

That is... holding down the multiple select modifier key the whole time:

- You click A, B, C. The renderer shows `[A,B,C]`, the internal model has `[A,B,C]`
- You click to desleect A. *Bug occurs* . The renderer shows `[ ]`, but the internal model is `[B, C]`
- If you click to select A again, the renderer now shows `[A]` but the internal model has `[A,B, C]`
- Click to select B again, and the internal model thinks you are *deselecting* B.  *Bug occurs again*. renderer shows `[]`, the internal model is now `[A, C]`. 

etc., etc.

https://github.com/user-attachments/assets/2d7935ae-7620-45b5-a1d1-ad041e29f0ab

Once you see this pattern, you can see it play out with edges.

https://github.com/user-attachments/assets/cf058df9-44c9-4475-94a8-3ab4858c2048

I hope this has meant something to someone!

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.12
  * Curves 0.6.50
```

depthoffocus, 2024-10-19:
Hard to show in a video but it also affects vertex selection the same way.

But when this bug happens, it only affects that selection type.

Deselecting a face only messes up the presentation of selected faces; the state of selected edges is unaffected. 

https://github.com/user-attachments/assets/35c3db2b-3a79-4271-9fce-302882039675

PaddleStroke, 2024-10-20:
I can reproduce the bug

PaddleStroke, 2024-10-20:
This is a quite serious regression as it introduce lot of confusion. Should probably be a blocker @maxwxyz

maxwxyz, 2024-10-20:
@PaddleStroke do you know a fix? IMO it is not as critical as crashes which we now consider blockers. Would be great to have in 1.0 but not sure if this would stop the release if this is the last issue. Otherwise for 1.0.1

PaddleStroke, 2024-10-20:
I've not looked into it yet. It does not crash but the confusion it introduce is big. And it concerns something super-basic which is selection. This is the kind of bug that would make freecad feels like crap for a 1.0 release. Much more than a crash that happens in very specific conditions imo.

depthoffocus, 2024-10-20:
A quick note: I'm just a random user and I don't know if it should be a blocker or not.

But I think what I found this evening is that because the bug isn't simple -- it's not just a question of losing your selections and having to start again, but the application being in an unpredictable state at that point -- it may be seriously frustrating to people who already have some difficulty selecting edges and especially vertices, which it also affects. 

They will have no idea why it is not working consistently, and as you will have a whole wave of people testing FreeCAD for the first time (or for the first time in a long time after abandoning it before), this is going to come up in a Youtube review.

maxwxyz, 2024-10-20:
I'm labeling it as blocker for further discussion. It seems it is just hiding the selection from the user as quickmeasure still has the result of the selection displayed.

depthoffocus, 2024-10-20:
Just to add one more observation:

If I run the` Gui.selection.removeSelection(...)` command that is injected into the python console log, the problem does not occur.

mwganson, 2024-10-20:
In MeshRemodel there is a go back selection (left green arrow icon) tool that can reset back to a prior selection state.   I noticed this bug while working on that function.  I mention here because it might be useful in illustrating / debugging the issue.

Create a default part workbench cube.  Select all of the faces.  I use Blender navigation, so I press Ctrl+Select to multiple select.  After selecting all the faces, Ctrl+Select one of them to deselect it.

Expected result: all faces are selected except for that one deselected face.
Apparent result: all faces are deselected.
Actual result: all faces appear to be deselected, but in face the selection is correct.

Click again in empty 3d space to clear the selection.  Then click the go back selection icon, select Block 1 as the option.  Result; the proper selection is shown.

The way the go back selection function works is it parses the python console and looks for those commented gui selection statements.  It uses those to build up the blocks of selections.  When if finds a clearSelection line it uses that as a delimiter for the blocks that get created.  One takeaway from this is the selection state that gets registered in the python console is correct.  I'm not sure if there is a user preference that affects whether the selections get echoed to the console as comments.  If there is a setting like that and the comments are turned off, then obviously this function will fail.

PaddleStroke, 2024-10-21:
If I have to take a guess I'd say it was introduced by this commit https://github.com/FreeCAD/FreeCAD/commit/85c1f5c47c651a60f14108281b22e0878c988a90
I'll start looking into this

PaddleStroke, 2024-10-21:
I can confirm that https://github.com/FreeCAD/FreeCAD/commit/85c1f5c47c651a60f14108281b22e0878c988a90 introduced this bug. Reverting the commit fix it. I'll study it to see how to keep this fix while fixing this issue as well.

closed 2024-10-21 by yorikvanhavre