FreeCAD/FreeCAD #16924: SIGSEGV when changing theme

berkakinci, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

2 out of 3 times, I got a crash when changing theme. I may have been editing a sketch at the time of crash.

There are other SIGSEGV in the forums, but I didn't find any mentioning this cause.

Reproduce:
- Have a document open (with a sketch open?)
- FreeCAD Menu
- Preferences
- General->Application->Theme

### Full version info

```shell
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

Originally reported in:
https://forum.freecad.org/viewtopic.php?t=91010
Crash Dump is in the thread.

Syres said: "The specific line that the Segfault is being tripped is [https://github.com/FreeCAD/FreeCAD/blob ... .cpp#L2378](https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Application.cpp#L2378) the QString styleSheetContent is present and appears to be correct but there could well be some references that it's not happy with possibly."

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berkakinci, 2024-10-02:
It looks like having an open sketch is the trigger:
I changed themes 3x with and open document w/ no active command: OK
I changed themes 1x with a sketch open in the same document: Crash

CalligaroV, 2024-10-06:
Hi @berkakinci!

I tried to reproduce the crash with
```ruby
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * freecad.gears 1.2.0
  * Beltrami 1.1.0
  * sheetmetal 0.4.24
  * Curves 0.6.44
  * fasteners 0.5.26
  * kicadStepUpMod 11.2.7
  * OpenTheme 2024.8.10
  * Render 2024.8.4
  * Ondsel-Lens 2024.7.5.02
``` 
in your same condition (with an open sketch, in edit mode and visible in the 3D View) but wasn't able.

Tried switching to various themes (OpenLight, OpenDark, FreeCAD Classic) and the only thing I see is a small lag while the theme is applied.
Also doing random clicks in the FreeCAD window has no effect, after the small lag everything looks normal with the updated theme.

By now, looking at the differences between your FreeCAD infos and mine the biggest difference is the OS (macOS for you, Linux for me).

Given that, AFAICS I would say that this is a macOS specific issue.

Feedback by other users (on various OSes) will surely help narrowing the scope of this issue.

zbynekwinkler, 2024-10-07:
I can reproduce the crash (with sketch open, added a rectangle to the xy-plane and then change theme).

```
OS: Debian GNU/Linux 12 (bookworm)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Debug
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

luzpaz, 2024-10-07:
Can't reproduce on 
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

luzpaz, 2024-10-07:
@johnocfii do you mind trying to reproduce on macos? TIA

JohnOCFII, 2024-10-07:
> @JohnOCFII do you mind trying to reproduce on macos? TIA

Boom!

Crashed first time through for me -- in my case attempting to switch from OpenLight to FreeCAD light.

It does **not** seem to crash moving FROM FreeCAD Light TO OpenLight.

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [9686]
Path:                  /Applications/Other Apps/FreeCAD-38871.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-07 10:30:36.9293 -0500
OS Version:            macOS 15.0.1 (24A348)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       01AB8A1D-6005-45CA-A51A-222A6B42A74F

Time Awake Since Boot: 100000 seconds
Time Since Wake:       377 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000004f02f6f3f30
Exception Codes:       0x0000000000000001, 0x000004f02f6f3f30

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [9686]

VM Region Info: 0x4f02f6f3f30 is not in any region.  Bytes after previous region: 4948598144817  Bytes before following region: 100123481784528
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  reserved VM address space (unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libQt5Widgets.5.15.13.dylib   	       0x10614f828 0x1060d0000 + 522280
1   libQt5Widgets.5.15.13.dylib   	       0x10613f4e0 0x1060d0000 + 455904
2   libFreeCADGui.dylib           	       0x10707fae0 Gui::Dialog::DlgSettingsGeneral::saveThemes() + 1104
3   libFreeCADGui.dylib           	       0x10707f2e8 Gui::Dialog::DlgSettingsGeneral::saveSettings() + 652
4   libFreeCADGui.dylib           	       0x10705df0c Gui::Dialog::DlgPreferencesImp::applyChanges() + 456
5   libQt5Core.5.15.13.dylib      	       0x1069ede84 0x1068a8000 + 1334916
6   libQt5Core.5.15.13.dylib      	       0x1069d1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
7   libQt5Core.5.15.13.dylib      	       0x1069d1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
8   CoreFoundation                	       0x1982a63a8 __CFRunLoopDoSource0 + 176
9   CoreFoundation                	       0x1982a610c __CFRunLoopDoSources0 + 244
10  CoreFoundation                	       0x1982a4cf8 __CFRunLoopRun + 840
11  CoreFoundation                	       0x1982a4334 CFRunLoopRunSpecific + 572
12  HIToolbox                     	       0x1a36dd0cc RunCurrentEventLoopInMode + 292
13  HIToolbox                     	       0x1a36e2d1c ReceiveNextEventCommon + 220
14  HIToolbox                     	       0x1a36e3020 _BlockUntilNextEventMatchingListInModeWithFilter + 76
15  AppKit                        	       0x19bde8a70 _DPSNextEvent + 660
16  AppKit                        	       0x19c70e7b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
17  AppKit                        	       0x19c016920 -[NSApplication _doModalLoop:peek:] + 216
18  AppKit                        	       0x19c321708 __33-[NSApplication runModalSession:]_block_invoke_2 + 84
19  AppKit                        	       0x19c321698 __33-[NSApplication runModalSession:]_block_invoke + 112
20  AppKit                        	       0x19c014e28 _NSTryRunModal + 100
21  AppKit                        	       0x19c321568 -[NSApplication runModalSession:] + 148
22  libqcocoa.dylib               	       0x10b30c3ac 0x10b2e0000 + 181164
23  libFreeCADGui.dylib           	       0x106f1466c Gui::Command::_invoke(int, bool) + 516
24  libFreeCADGui.dylib           	       0x106f1442c Gui::Command::invoke(int, Gui::Command::TriggerSource) + 348
25  libQt5Core.5.15.13.dylib      	       0x1069ede84 0x1068a8000 + 1334916
26  libQt5Core.5.15.13.dylib      	       0x1069d1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
27  CoreFoundation                	       0x1982a63a8 __CFRunLoopDoSource0 + 176
28  CoreFoundation                	       0x1982a610c __CFRunLoopDoSources0 + 244
29  CoreFoundation                	       0x1982a4cf8 __CFRunLoopRun + 840
30  CoreFoundation                	       0x1982a4334 CFRunLoopRunSpecific + 572
31  HIToolbox                     	       0x1a36dd0cc RunCurrentEventLoopInMode + 292
32  HIToolbox                     	       0x1a36e2d1c ReceiveNextEventCommon + 220
33  HIToolbox                     	       0x1a36e3020 _BlockUntilNextEventMatchingListInModeWithFilter + 76
34  AppKit                        	       0x19bde8a70 _DPSNextEvent + 660
35  AppKit                        	       0x19c70e7b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
36  AppKit                        	       0x19bddbb7c -[NSApplication run] + 480
37  libqcocoa.dylib               	       0x10b30c4d0 0x10b2e0000 + 181456
38  freecad                       	       0x1049c7184 main + 5128
39  dyld                          	       0x197e3c274 start + 2840
```

berkakinci, 2024-10-07:
I’m not sure how important it is to try reproduce the issue on different OS. Issues like use-after-free (speculating) are not reliably reproducible anyway.
There are tools for pinpointing these issues.  I can’t remember if this was a build option or command line option or external tool like Valgrind.  Someone more familiar with FreeCAD debugging should guide us.  You’re better off reproducing with those tools enabled.

sliptonic, 2024-10-07:
The specific combination of events necessary to see the segfault is pretty unlikely.  We're leaving this as 1.0 milestone but not making it a blocker for 1.0 release.
Definitely should be fixed but we won't hold up 1.0 for it.

berkakinci, 2024-10-07:
I disagree. It was one of the first things I tried on the new version — when I noticed sketcher geometry visibility was poor. BTW, it’s poor in all three themes, but _that_ is definitely not a release stopper.

JohnOCFII, 2024-10-07:
@MisterMakerNL  -  Is this in your wheelhouse?

MisterMakerNL, 2024-10-07:
I would think Qtstyle is the reason. But @Syres916 says it's the stylesheet and qtstyle is set in the preference pack.
Could we test this on old releases, and maybe on a QT 6 build?

Syres916, 2024-10-07:
Using EndeavorOS (Qt6) I couldn't even change themes 3 times, segfaulted on the 2nd, though it doesn't have as much memory as the Qt5 box I initially did the test last week on.

```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38881 (Git)
Build type: Release
Branch: main
Hash: ab44bd1f7ba3c3008f3fd8ef6bdb61d11d1a82e1
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
```

berkakinci, 2024-10-07:
> Found this mention of the tool I’m talking about: https://forum.freecad.org/viewtopic.php?t=52775&start=30#p560243

I reiterate this. If you are going to dig in to the  issue, try the special build to reproduce.  Otherwise, you’re possibly wasting your time.

MisterMakerNL, 2024-10-08:
Could this be a thing https://forum.qt.io/post/661460 ?

Syres916, 2024-10-08:
> Could this be a thing https://forum.qt.io/post/661460 ?

I don't think it's relevant here but I'll know more when I followed up on a lead Git Bisect has given regarding the new Materials. When the user clicks `OK` to the _Preferences_ dialog there isn't just the Stylesheet change being analysed but things like _Shape Appearance_ are also reviewed/updated which is where my investigation is right now. It's going to be probably another 24 hours before I give an update as there's another large Git Bisect to do.

Syres916, 2024-10-09:
I'm not sure _how_ the material changes and `QtStyle` are linked but that's the root cause of the Seg fault. I'm looking to simply make the `Preferences` > `General` > `Theme` combobox/Pref Pack _Apply_ buttons and Start Theme selector be set to disabled when Sketcher Edit Mode is active as a initial first step. What I also need to ensure is that other specific areas such as Fem/CAM new simulator aren't affected.

Syres916, 2024-10-09:
Once I've cleared the CI tests and fixed any Lint feedback I'll submit the PR so hopefully this evening.

berkakinci, 2024-10-11:
Should this bug survive in less severe form?  As understand it, the root cause isn’t fixed; but, the user is not allowed to cause it any more.

luzpaz, 2025-08-30:
@berkakinci can you test on latest dev build when a new build is generated (**after** 8/27) https://github.com/FreeCAD/FreeCAD/releases

berkakinci, 2025-08-30:
I will try when the next build is out.