FreeCAD/FreeCAD #16926: Possibility to split the configuration folder depending on the program version.

kaiwas, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

At the moment, the configuration folder is ///FreeCAD (the beginning of the path differs depending on the operating system). At the same time, users can simultaneously use versions 0.21, 0.19 and 0.22
Due to the fact that each version has its own features, problems often arise (for example, due to the removal of the Arch workbench)
I have repeatedly noticed topics on the forum like this.
https://forum.freecad.org/viewtopic.php?t=91042

And I myself encountered problems when running version 0.21 (having previously configured 0.22)

Maybe it is worth introducing a rule for the configuration folder that would work only for a specific version?

After all, version 1.0 will be followed by version 1.1, 2.0, and so on, and changes in them may be incompatible with previous versions.
Maybe ///FreeCad_1.0 ///FreeCad 1.1 and so on.

A good example is the pod in blender (again). There, a separate configuration folder is created for each key version, and when you first run a new version, you are offered to transfer the settings from the previous version. After the transfer, we can do anything in the configuration without worrying that the previous version of the program will break.

[image]

I know that the ability to separate the configuration folder is available for console commands or creating launch rules in .bet. But you need to be able to do such manipulations and know the rules. I think that an ordinary user would not want to deal with this.

### Full version info

```shell
any versions
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GeneGH, 2024-09-30:
The single configuration file has been a headache for a long, long time. Yes, there are ways around it, but this change (or some version of it) would be a great plus.

luzpaz, 2024-10-15:
Related tangentially to #6106