FreeCAD/FreeCAD #16929: Custom colors are not getting saved

leoheck, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have to constantly add the same colors to the custom colors.. 

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git) AppImage
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-09-30:
I cannot reproduce this. @leoheck, where do you open the color dialog. Please give steps to reproduce.

leoheck, 2024-09-30:
@benj5378 I open it from the context menu most of the times or using Ctrl+D.

Also, if you see my imagine it has 4 custom colors (top-right). Those were selected some time/versions ago. Recently colors I am adding there are not staying there anymore.

Steps to reproduce?
- Select an object 
- Ctrl+D
- Go to the button that allows to customize the color. 
- Select the diffuse color 
- Then choose a color (Drag or Add custom color)
- Close the panel...
- Close the file...
- Open it again...
- Check the colors gone.

Syres916, 2024-09-30:
On my Linux Mint 22 box the Custom Colors are saved correctly in /home/_username_/.config/QtProject/QtProject.conf and as this is not coded in FreeCAD directly (it's Qt) then it's not surprising that it works the same in 0.21.x as it does using the most recent main branch. So I suspected this maybe an AppImage specific issue but no it works correctly updating the same file as my self-compiled builds. Conclusion: cannot replicate issue with any build.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.3.33846 (Git)
Build type: Release
Branch: 021_Python_3_12_And_Other_Fixes
Hash: 7e1da2e96b108a5b8f95b88a7eb431bedf04db61
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```

kaiwas, 2024-10-01:
Reproduced the sequence of actions. Colors are preserved. (Not Confirm)

leoheck, 2024-10-01:
I had to add a single color 6x times already.
When the color is saved? It should be saved as sono as It is added to the grid of custom colors.

leoheck, 2024-10-01:
It is not saving the color.

leoheck, 2024-10-01:
> once again this is absolutely nothing to do with FreeCAD

Ah, you already know about this bug then?

benj5378, 2024-10-01:
@luzpaz, can we close this as either being a 3rd party issue, that FreeCAD has no control over, or intended behaviour?

leoheck, 2024-10-01:
It is a Freecad bug. Ben5373 is supposing it is a 3rd party bug since he could not reproduce the problem. But he didn't point any related known bug on the 3rd party lib. This doesn't mean the FreeCAD doesn't have the issue. It shows I don't have the precise steps to reproduce the issue yet or he is not following the steps correctly.

leoheck, 2024-10-01:
@kaiwas I see you reacted with thumbs up. Are you observing the same issue?

benj5378, 2024-10-01:
> It is a Freecad bug. Ben5373 is supposing it is a 3rd party bug since he could not reproduce the problem. But he didn't point any related known bug on the 3rd party lib. This doesn't mean the [...]

Dude. I'm basing it on what @Syres916 said, that it's QtColorDialog made by Qt and not by FreeCAD: https://doc.qt.io/qt-6/qcolordialog.html. If it is the case, that the QtColorDialog doesn't save colors, then you have to report it to Qt who made the dialog on https://bugreports.qt.io

kaiwas, 2024-10-01:
> I see you reacted with thumbs up. Are you observing the same issue?

No. I couldn't reproduce this problem (everything is fine with me).
I've previously encountered the problem that colors weren't applied when I clicked the "OK" button (without adding to the palette), but it seems that this problem has been solved too.

If you want, I'll remove my thumb up. I think I put it there by mistake.

luzpaz, 2024-10-05:
Can someone locate an upstream Qt bug for this issue?

benj5378, 2024-10-05:
@luzpaz, I don't even think there is a bug here. Seems like intended behaviour

leoheck, 2024-10-05:
How is the intended behavior to have custom colors reset every time? Perhaps you didn't understand the issue. The issue is that colors are not persisting after putting them on the grid of custom colors.

luzpaz, 2024-10-06:
From an intuitive UI perspective it makes sense that the chosen colors should persist instead of getting reset each time. So I'm in agreement with @leoheck that this shouldn't be intended behavior.

benj5378, 2024-10-06:
> the chosen colors should persist instead of getting reset each time

@luzpaz, but is this really what is happening? No one could reproduce this, and @Syres916 pointed out, that it's due to the missing step of pressing OK button.

Let me know, if anyone choosing a color on the pallete **and** presses OK, doesn't have their colors saved.

leoheck, 2024-10-06:
I will try to reproduce it again here, now. Let me give you right instructions. Otherwise I am going to close this.

leoheck, 2024-10-06:
Alright, I am dragging the color to the first position, hitting OK, so save/close the popup... saving and closing the file. Opening it again, the color is there, but when I launch the color selector it looks like it is resting something. Please, check this video. And unfortunately I could not display the cursor.

[Screencast from 2024-10-06 11-32-32.webm](https://github.com/user-attachments/assets/9577af86-53a5-417a-b34f-4e12ff0c8801)

leoheck, 2024-10-06:
I am launching the color widget with the Ctrl+D on the Body or in the last feature. Does not matter, both should have the saved color available for me.

leoheck, 2024-10-27:
This is still broken and now, just clicking the `Custom Appearance` button sometimes reset the color of the object. Very weird.

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

luzpaz, 2024-10-28:
@leoheck would you consider testing the flatpak beta (which uses Qt6) ?

leoheck, 2024-10-28:
I am starting to think this settings is saving colors per model. So if I have 10 Bodies in the same file, that have to use the same Custom color... I will have to save the custom color for each Body which seems useless, since it is the same as copying pasting the color code on each Body.

xtemp09, 2024-10-28:
@leoheck, can you post the output of _env_ command? Do you use options/scripts/environment variables to launch _FreeCAD_ AppImage?

leoheck, 2024-10-28:
@xtemp09 , sure, here it is.

```env
SYSTEMD_EXEC_PID=4659
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
SESSION_MANAGER=local/falcon:@/tmp/.ICE-unix/4659,unix/falcon:/tmp/.ICE-unix/4659
LANG=en_US.UTF-8
LOGNAME=lheck
XDG_CURRENT_DESKTOP=ubuntu:GNOME
COLORTERM=truecolor
XDG_RUNTIME_DIR=/run/user/1000
TILIX_ID=90278a4d-b76f-4654-be43-ce32ff129ce8
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
LIBVIRT_DEFAULT_URI=qemu:///system
DESKTOP_SESSION=ubuntu-xorg
QT_IM_MODULE=ibus
XDG_MENU_PREFIX=gnome-
USER=lheck
HOME=/home/lheck
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus,guid=7dd7e34cf2de45c78ebc214e671e839b
GTK_MODULES=gail:atk-bridge
XDG_DATA_DIRS=/usr/share/ubuntu-xorg:/usr/share/gnome:/home/lheck/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
WINDOWPATH=2
JOURNAL_STREAM=8:36203
GSM_SKIP_SSH_AGENT_WORKAROUND=true
QTWEBENGINE_DICTIONARIES_PATH=/usr/share/hunspell-bdic/
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
MANAGERPID=4156
QT_ACCESSIBILITY=1
XDG_SESSION_DESKTOP=ubuntu-xorg
DEBUGINFOD_URLS=https://debuginfod.ubuntu.com 
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu-xorg:/etc/xdg
VTE_VERSION=7600
GNOME_SHELL_SESSION_MODE=ubuntu
MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA=
XMODIFIERS=@im=ibus
SHELL=/usr/bin/zsh
XDG_SESSION_TYPE=x11
DBUS_STARTER_BUS_TYPE=session
PATH=/home/lheck/Documents/stp2gltf:/home/lheck/Dropbox/Documents4/kiri/bin:/home/lheck/Dropbox/Documents4/kiri/submodules/KiCad-Diff/bin:/home/lheck/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/lheck/Applications:/home/lheck/.cargo/bin:/home/lheck/.roswell/bin
MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/app.slice/app-gnome\x2dsession\x2dmanager.slice/gnome-session-manager@ubuntu.service/memory.pressure
USERNAME=lheck
INVOCATION_ID=1ca196f2f67e4e0bb9c82599e71d3e89
PWD=/home/lheck
XAUTHORITY=/run/user/1000/gdm/Xauthority
QT_QPA_PLATFORMTHEME=qt5ct
DBUS_STARTER_ADDRESS=unix:path=/run/user/1000/bus,guid=7dd7e34cf2de45c78ebc214e671e839b
DISPLAY=:0
GDMSESSION=ubuntu-xorg
SHLVL=1
XDG_SESSION_CLASS=user
TERM=xterm-256color
OLDPWD=/home/lheck
ZSH=/home/lheck/.oh-my-zsh
PAGER=less
LESS=-R
LSCOLORS=Gxfxcxdxbxegedabagacad
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.crdownload=00;90:*.dpkg-dist=00;90:*.dpkg-new=00;90:*.dpkg-old=00;90:*.dpkg-tmp=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:*.swp=00;90:*.tmp=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:
SPACESHIP_VERSION=4.16.0
SPACESHIP_ROOT=/home/lheck/.oh-my-zsh/custom/themes/spaceship-prompt
OSH_AUTODOC_PDF_RES_DIR=/home/lheck/.roswell/local-projects/osh-autodoc-pdf/res
_=/usr/bin/env

```

xtemp09, 2024-10-28:
> ```ruby
> OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
> ```

> ```dotenv
> QT_QPA_PLATFORMTHEME=qt5ct
> ```

From here: https://sourceforge.net/p/qt5ct/code/HEAD/tree/tags/qt5ct-1.8/

```
Files and Directories:

qt5ct - Qt5 configuration tool
libqt5ct.so - qt5ct platform plugin
libqt5ct-style.so - qt5ct proxy style plugin
libqt5ct-common.so - qt5ct shared library
~/.config/qt5ct/qt5ct.conf - configuration file
~/.config/qt5ct/qss, ~/.local/share/qt5ct/qss, /usr/share/qt5ct/qss, /usr/local/share/qt5ct/qss - style sheets
~/.config/qt5ct/colors,~/.local/share/qt5ct/colors, /usr/share/qt5ct/colors,
/usr/local/share/qt5ct/colors - color schemes
/etc/xdg/qt5ct/qt5ct.conf - default configuration file
```

Can you look at these directories and play with `QT_QPA_PLATFORMTHEME`?

leoheck, 2024-10-28:
@xtemp09 I didn't understand. Are you saying this qt5ct is breaking things?

xtemp09, 2024-10-28:
> @xtemp09 I didn't understand. Are you saying this qt5ct is breaking things?

I don't know the cause of the problem. Most likely, qt5ct does not, `QT_QPA_PLATFORMTHEME` does.

Does your `~/.config/QtProject.conf` have `-rw-rw-r--` attributes? Does it change during execution of _FreeCAD_? Try to temporarily remove `QT_QPA_PLATFORMTHEME` and see if the problem persists.

leoheck, 2024-10-28:
I purged `qt5ct` to check this, I definitely dont need it. I also removed `~/.config/Qt*` (sorry too late). I still have to check this when I get time, it will have to be later today.

I also dont need this theme `QT_QPA_PLATFORMTHEME`, so I am going to unset this.

xtemp09, 2024-11-10:
@leoheck, does it work?

xtemp09, 2024-12-03:
@leoheck, 👆

luzpaz, 2024-12-04:
https://bugreports.qt.io/browse/QDS-22?jql=text%20~%20%22color%20picker%22 is it one of these ?

luzpaz, 2024-12-05:
> I can tell you later

@leoheck any progress ?

Aside, we should really start offering Qt6 builds for testing.

luzpaz, 2025-07-28:
cc @leoheck can you retest with latest 1.1-dev built with qt6 ?

leoheck, 2025-07-30:
Hey @luzpaz, I haven't noticed this issue anymore, and while testing it just now, I couldn't reproduce it.