FreeCAD/FreeCAD #16931: PartDesign: Polar pattern over boolean cut operation is not done properly

hasecilu, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to make a spacer for a 3D printed bearing, in the body that makes a bearing separator cage the boolean cut operation is not properly done but the fuse operation is done.

The first cut is done properly but the pattern not.
[image]

### cut
[image]

### fuse
[image]

### Full version info

```shell
OS: Arch Linux (Hyprland/hyprland)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git) AppImage
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Mexico (es_MX)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * ThreadProfile 1.92.0
  * FreeCAD-Movie 2024.2.15
  * freecad.gears 1.2.0
  * PieMenu 1.8.0
  * freecad_metal_workbench 0.0.1
  * Ship 2022.4.11
  * OpenTheme 2024.9.1
  * FreeCAD_WeldFeature 0.1.0
  * FreeGrid 2.0.0
  * FreeCAD-Reinforcement
  * Defeaturing 1.2.2
  * FreeCAD_SheetMetal 0.4.26
  * FreeCAD-Pyramids-and-Polyhedrons
  * yaml-workspace 0.1.2
  * woodworking 0.21.2.33771
  * GDML 2.0.0
  * Assembly4 0.50.10
  * FastenersWB 0.5.29
  * lattice2 1.0.0
  * Rocket 3.3.0
  * dodo 1.0.1
  * DidYouKnowThat
  * FreeCAD-motion
  * cadquery_module
  * Profiles 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * osh-autodoc-workbench 0.2.3
  * CfdOF 1.27.10
  * SteelColumn
  * A2plus 0.4.67
  * kicadStepUpMod 11.2.4
  * frame 0.1.1
  * Curves 0.6.47
  * FreeCAD_assembly3 0.12.2
  * Render 2024.9.25
```

### Subproject(s) affected?

PartDesign

### Anything else?

My file, change the extension, don't extract it:
[Bearings.zip](https://github.com/user-attachments/files/17194768/Bearings.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Matou25, 2024-09-30:
same kind of issue with polor patern
i create a pocket with polyline then add fillet on angle when i use polar patern, surface is not refine properly execpect for the orignale pocket. see below :
[image]
[image]

try with 1.0 RC1 and 1.0 RC2

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Assembly3 0.12.2
  * BillOfMaterials 0.1.4.2
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * kicadStepUpMod 11.3.2
  * sheetmetal 0.4.25
  * WebTools 1.0.0

mwganson, 2024-09-30:
Boolean objects are not intended for use as patterns shapes for patterning tools, such as polar pattern.  Use instead the subtractive sphere to make the cut, then you should be able to polar pattern the subtractive sphere.

hasecilu, 2024-09-30:
> Boolean objects are not intended for use as patterns shapes for patterning tools, such as polar pattern. Use instead the subtractive sphere to make the cut, then you should be able to polar pattern the subtractive sphere.

That was the issue, thanks for the help.

closed 2024-09-30 by hasecilu