FreeCAD/FreeCAD #16933: Part: Part_Section: Section of two Shapes - Extruded Sketch with Plane - Wire not closed

AIRCAP, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When cutting a Part_Section (Section of two Shapes) of an extruded part with anything but the most primitive shapes, the section is not "closed" - meaning it cannot be used in most tools - for example for further extrusions or to enclose a surface.

[image]

Curiously enough when "Draft_Draft2Sketch" is applied to this section, the resulting sketch will be closed, so the knots must be off by less by miniscule amounts.

Example attached:
[part_section_bug.zip](https://github.com/user-attachments/files/17194941/part_section_bug.zip)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38844 (Git)
Build type: Unknown
Branch: main
Hash: 6e2cd4e73303f817e4bf5e908abe39030b7c5fb3
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-10-02:
There is a sketcher bug where you must sometimes wiggle edges to get the wire "closed".  This might be related internally.  I suspect an issue with the Part.Wire.isClosed() method.  I used the OpenEdgeFinder tool from MeshRemodel workbench to check this wire.  At default OCCT tolerance of 1e-7 it found no open edges.  I had to go all the way down to 1e-12 to find an open edge.  The gap could have been over 4500 times bigger and still would have been within the stated 1e-7 tolerance range.

Part.makeFilledFace works to make a face.  And when checking the face it shows to be planar and the face.OuterWire shows to be closed.  Here is a small macro to create a parametric face from these objects:

```
import Part, FreeCAD, FreeCADGui

class SectionFace:
    def __init__(self, obj):
        obj.Proxy = self
        obj.ViewObject.Proxy = 0
        grp = "SectionFace"
        obj.addProperty("App::PropertyLink","Section",grp,"Section object to make a face out of")
        
    def execute(self,fp):
        if not fp.Section:
            fp.Shape = Part.Shape()
            return
        surf = Part.makeFilledFace(fp.Section.Shape.Edges)
        fp.Shape = surf
if __name__ == "__main__":
    doc = FreeCAD.ActiveDocument
    import SectionFace
    import importlib
    importlib.reload(SectionFace)
    if doc:
        fp = doc.addObject("Part::FeaturePython","SectionFace")
        SectionFace.SectionFace(fp)
        sel = FreeCADGui.Selection.getSelection()
        if len(sel) == 1:
            if sel[0].isDerivedFrom("Part::Feature"):
                fp.Section = sel[0]
        doc.recompute()
    else:
        FreeCAD.Console.PrintError("No active document.\n")
```

Save the macro file as SectionFace.py  (not SectionFace.FCMacro or else it won't work properly) and save it to your macro folder.  Select the section object and run the macro or else set the Section property to the Section object.

maxwxyz, 2024-10-05:
Is this an issue in the stable 0.21.2 version or a regression?

FEA-eng, 2026-09-07:
The file fails to recompute and still shows "Wire not closed"

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```