FreeCAD/FreeCAD #16934: Recompute Error when opening file created with v0.21.2 in v1.0.0RC2 (SubtractiveLoft: Loft: Failed to create shell)

mattj518, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I just finished creating a simple part with v0.21.2 and decided to try to create it in the new version to try it out. I was unable to get the SubtractiveLoft to work as it did in 0.21.2. I then tried to open the file created in 0.21.2 and received a "Recompute Error". The Report View listed the following:
11:04:17  SubtractiveLoft: Loft: Failed to create shell
11:17:17  <Exception> FeatureDressUp.cpp(209): Invalid edge link: Edge254
11:17:17  Chamfer: Invalid edge link: Edge254

I'm somewhat new to FreeCAD so I may be doing something incorrectly, but since this worked in v0.21.2 and not in v1.0.0RC2, I decided to report it.

### Full version info

```shell
OS: KDE neon 6.1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

None

### Anything else?

[Alignment Tool.zip](https://github.com/user-attachments/files/17195640/Alignment.Tool.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-05:
@FlachyJoe could this be refine related? Or is this a TNP change?

FlachyJoe, 2024-10-05:
@maxwxyz Currently Refine is switch on at object creation and the property state is saved. So an object saved with Refine false is not refined even when opened in a system with auto-refine preference is on.

Even in 0.21.2 the given file is broken:
[image]

maxwxyz, 2024-10-05:
I'm closing this then if also broken in the stable version.
@mattj518 feel free to add more info and reopen if there is an issue.

closed 2024-10-05 by maxwxyz