FreeCAD/FreeCAD #16935: SVG geometry import can break open closed paths

Scafir, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In certain cases, it seems like the svg import of freecad breaks paths and opens them very slightly.
[image] is the most minimal example producing the issue that I could produce.
It is also worth noting that, at the zoom level required to see the open path, it is near impossible to select nodes and vertices.

Thank you for the good work.

### Full version info

```shell
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/ProDark/Qt default
Installed mods: 
  * kicadStepUpMod 11.2.9
  * 3D_Printing_Tools
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-30:
Attachment doesn't display.

Scafir, 2024-10-01:
Hi, this is normal, as the line segments are very small and inkscape (used for producing the svg) was using A4 size.
I have resized the view, and edited the issue with the file as an attachement rather than image: even with the resized view, the path is barely visible.

Scafir, 2024-10-06:
I had a bit of time to try and look into it more, here is what I found so far:
* The concerned file is `src/Mod/Draft/importSVG.py`, in `svgHandler.startElement()`.
* Removing any set of coordinates in the svg data results in a closed path.

I was able to shrink down the minimal example [more](https://github.com/user-attachments/assets/acb3bdca-c398-4b12-92d9-cf935d565511).

Scafir, 2024-10-12:
I think I now know why this is happening.

When creating an svg path, the parser acts like a pen on paper: it starts at point A, and continues through B, C, etc.
These moves can be relative and hence, very small tolerances error can build up during the creation of the path, resulting in a path defining a closed shape having ever so slightly different start and end points. This inconsistency is not really a problem from an svg point of view, as closed path are terminated with the "z" command, indicating that the past must be closed between start and end point no matter what.

Where it breaks down is in how these are considered when parsing the SVG. It seems that, for closed path to be created, this start and end point must be exactly the same.
How it is achieved in the code currently is that we check if the start and end vertices are the same. If they are not the same, then we close the path with a straight vector between the two:
https://github.com/FreeCAD/FreeCAD/blob/e478192aa5393de9b7df190341a043f79ec6ac49/src/Mod/Draft/importSVG.py#L1209-L1216

It however seems that the command used to check the point equivalence `DraftVecUtils.equals(lastvec, firstvec)` does not use the same tolerances than than `Part.Wire`, used to create the closed path. Thus, it seems that there is a small window where the parser considers the start and end point to be the same, when they actually differ enough that `Part.Wire` creates a open path, rather than closed.
Removing the check and always creating an extra edge to close the path results in my example file being parsed correctly.

There is thus two solutions:
1) Use an equality function that matches `Part.Wire` behavior.
2) Always close the path with a straight edge (what if the edge is of length zero?)

Any advice from somebody that knows more about freecad's code? This is my very first experience with it.

luzpaz, 2024-10-13:
cc @yorikvanhavre

yorikvanhavre, 2024-10-15:
I would not remove that test because `Part.LineSegment(lastvec, firstvec).toShape()` will throw an error if both points are equal.

What we could do instead, is use an absolute comparison, and instead of:

```python
if not DraftVecUtils.equals(lastvec, firstvec):
```

do:
```python
if lastvec != firstvec:
```

Scafir, 2024-10-29:
Thank you for the help @yorikvanhavre @luzpaz  !

I spent more time than I'd care to admit on this. It can get very confusing because different parts may have different precision, such as the python float type, the svg standard, `DraftVec`, `Vector`, `Part`.
After quite a lot of investigation, I was not able to determine how exactly FreeCAD determines if a `Part`'s path is closed.
Doing an absolute comparison does not seem to be a perfect solution, as `Vector(0.0,0.0,0.0) == Vector(1e-100,0,0)` returns true, but `a =  Part.LineSegment(Vector(0.0,0.0,0.0), Vector(0.0,0.0,1e-100))` does not fail (weird, no?). However, when crafting an svg that would reproduce the issue, the path is detected as closed. 

As these precision considerations feel quite far out of my understanding of FreeCAD, I think that we could go either go ahead with the absolute comparison suggested by yorikvanhavre. It's also interesting to point that as the DraftVec precision is used all over the SVG importer, it is not possible to use it for very small (nanometer) designs.

I will try to make a pull request over the coming days with this modification.

FEA-eng, 2026-01-11:
> zoom level required to see the open path

Can you share a screenshot? I'm not sure what to look for in those files.

FEA-eng, 2026-10-12:
See [this](https://github.com/FreeCAD/FreeCAD/issues/16935#issuecomment-3734665379)