FreeCAD/FreeCAD #16937: Recompute Error when opening file created with v0.21.2 in v1.0.0RC2 (Curves: ordon fails to recompute)

EarlJay1, 2024-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On recomputing the model when opening the 1.0RC for the first time, it fails reporting errors regarding the Gordon surface.
The AddOns were updated.
Don't know if this goes into here, but the file opens fine in 0.21.2

2:22:56  pyException: Traceback (most recent call last):
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordonFP.py", line 144, in execute
    s = gordon_surf.surface()
        ^^^^^^^^^^^^^^^^^^^^^
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 358, in surface
    self.perform()
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 326, in perform
    self.gordon_surf = builder.surface_gordon()
                       ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 109, in surface_gordon
    self.perform()
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 105, in perform
    self.create_gordon_surface()
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/gordon.py", line 230, in create_gordon_surface
    surfaces_vector = bsa.createCommonKnotsVectorSurface(surfaces_vector_unmod, self.par_tol)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/BSplineAlgorithms.py", line 363, in createCommonKnotsVectorSurface
    self.makeGeometryCompatibleImpl(adapterSplines, tol)
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/BSplineAlgorithms.py", line 336, in makeGeometryCompatibleImpl
    spline.insertKnot(resultKnots[knotIdx], resultMults[knotIdx], par_tolerance)
  File "/home/earl/.local/share/FreeCAD/Mod/Curves/./freecad/Curves/BSplineAlgorithms.py", line 104, in insertKnot
    raise RuntimeError
<class 'RuntimeError'>: 

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Curves 0.6.48
  * fasteners 0.5.29
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-03:
I'd propose to open this issue in the Curves repository. Feel free to reopen if this is an issue in FreeCAD.

closed 2024-10-03 by maxwxyz

EarlJay1, 2024-10-03:
@maxwxyz 
Done, but I doubt that anything there will happen. Opened another issue in June in the Curves repo and there wasn't even a reaction until now.