FreeCAD/FreeCAD #16940: The delimiter in the exported CSV file is the TAB character in Assembly/Spreadsheet workbench, but shouldn't it be a comma?

H-Tsubota, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I used the "Create Bill of Materials" function in the Assembly Workbench to export a BOM as a CSV file, but the delimiter in the file was the TAB character.

Since it is a CSV file (Comma Separated Values), shouldn't the delimiter be a comma?

After examining the source code, I found that the Assembly workbench's BomObject class inherits from Spreadsheet::Sheet class, and the delimiter for the CSV file export is controlled by user preferences, with the default being the TAB character.

When I changed the delimiter to a comma in the Spreadsheet workbench's user preferences dialog, the delimiter also became a comma on export of the BOM as a CSV file in the Assembly workbench.

Shouldn't the default delimiter be a comma in the Spreadsheet user preferences, or shouldn't the delimiter be a comma only in the Assembly workbench?

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-01:
This has been reported before, this is just one discussion on the forum: https://forum.freecad.org/viewtopic.php?style=10&t=56333

I agree, that this is an issue as the name CSV kind of implies comma to be used. But using tabs as delimiters are not technically wrong.

CSV is a hot mess, and it has never been standardized as the format is so simple and became popular before any standard was written for it. The closest to a standard that I've found is the specification RFC 4180, that was written in 2005.

My guess is that the current default could be due to many counties use comma as decimal place and this could historically have caused problems.

Changing the default feels wrong though, this will most certainly cause issues for many current users who is used to the current behavior.

Potential solution:

* Show a dialog when importing/exporting to csv.
* Let the user specify which delimiter to use (radio buttons)
* Let the setting control which option is the default.
* Make sure export/import aligns with RFC 4180.

H-Tsubota, 2024-10-01:
Thanks for your comment. I seem to have missed that forum post. Sorry.

The biggest problem is that Assembly workbench users will not find a way to change the setting, because most users don't want to read the source code.

I agree that changing the default would confuse current users, but I think this moment, at version 1.0, is the best chance to break backward compatibility.

Personally, I think it would be easier for users to understand if they selected ".csv" in the file save dialog, which would save the file separated by commas, and if they selected ".tsv", which would save the file separated by tabs.

As seen in the IANA MIME type standard, there seems to be a clear distinction between .csv format and .tsv format.

Roy-043, 2024-10-01:
There is already a preference for this.
RTM: https://wiki.freecad.org/Spreadsheet_Preferences

H-Tsubota, 2024-10-01:
To hyarion:

> My experience with .csv files is that many issues starts with assumptions.

You're absolutely right.

For reference, I did some research on the major spreadsheet software. Since these apps are widely used, I think that at least these specifications would not cause too much confusion.

**Excel (Excel 2013)**
- .txt ---> TAB separated
- .csv ---> comma separated
- .prn ---> space separated

In .csv, the delimiter can be changed to any character in the user settings.

**Google Sheets**
- .csv ---> comma separated
- .tsv ---> TAB separated

**LibreOffice Calc (LibreOffice 24.8.2)**
- .csv ---> Extra export settings dialog is poped in first exporting and a user set the delimiter (default is comma).

Anyway I respect your experience.

I'd like to dig into the FreeCAD source code a bit to see what specifications would be preferable.

To Roy-043:

I know it (please read my first post).

The points are that Assembly workbench users may not see the Spreadsheet workbench settings and/or the confusion comes from the fact that the default delimiter for CSV files is TAB.

But thanks for your comment.

mwganson, 2024-10-01:
For locales that use comma for decimal, obviously a comma as the delimiter is not going to work.  Therefore, the comma cannot be the default.  I personally have the comma as the separator, but I'm not in one those comma = decimal locales.

There is a help button in the BOM export dialog.  It would be a simple fix to put a note in there that the spreadsheet CSV export settings will be also used for this export.

hyarion, 2024-10-02:
@mwganson This is unfortunately a common misconception (that also could be the reason why FreeCAD's .csv files uses tabs), as RFC 4180 handles that case quite clearly.

From RFC 4180, (_with my own comments, not part of the RFC, in italic_):
1. Each field may or may not be enclosed in double quotes (Section 2, paragraph 5) - _There's no difference between `1337` and `"1337"`. Fields do not have a type, so having double quotes doesn't differentiate strings from numbers as in other text file formats_.
2. Fields containing line breaks (CRLF), double quotes, and commas should be enclosed in double-quotes (Section 2, paragraph 5) - _A field that store the information `3,14` is stored as `"3,14"`_.

Roy-043, 2024-10-02:
In CSV files created by the Spreadsheet WB a point is always used as the decimal separator. The current locale is ignored.

H-Tsubota, 2024-10-08:
To improve the problem with minimal changes, I propose the following three changes.

- patch1

Added a description of how to customize the format in the BOM exporting of the Assembly workbench.

- patch2

Change the CSV export function of the Spreadsheet workbench to enclose field values ​​with quotation characters if they containing delimiters.

When the delimiter is a comma, this change makes that RFC4180 section 2.6 is partially satisfied.

- patch3

Changed the default delimiter to a comma in the Spreadsheet workbench settings.

Patch1 and patch2 do not change the current default behavior (in TAB-delimited CSV).

Patch3 will obviously be controversial, because there are FreeCA users who support "TAB-delimited CSV files" and those who support "comma-delimited CSV files". The "TAB-delimited CSV files" supporters are concerned about the confusion that will arise in countries where a comma is used as the decimal point. However, I think that such confusion can be avoided to some extent by patch2.

I have already submitted a pull request for patch1 (#17091), which has few side effects.

If there are no objections, I plan to submit a pull request for patch2 and patch3.

Please let me know if you have any opinions.

hyarion, 2024-10-08:
In my opinion, import and export (from the gui) needs to have a dialog where these settings can be configured easily by the user.

This is a common pattern for import/export which is also used for STEP import/export (I don't think anyone will be annoyed by seeing this the first time). I think this will solve most issues, with as little annoyance as possible.

I think good UX could eliminate the issue completely rather than writing documentation for how to work around the issue (patch 1).

> Change the CSV export function of the Spreadsheet workbench to enclose field values ​​with quotation characters if they containing delimiters.

This is a good idea. But maybe we should swap out the import/export functionality to use python's CSV module while we're at it. Python's CSV module has been battle tested and we wouldn't need to reinvent the wheel ourselves (plus we wouldn't need to maintain it ourselves).

Regarding patch 3: If we create an export dialog, and make sure that upgrading users get their old setting, then I don't see any issue with changing the default for new installations.

---

I'm almost ready to present an _import_ dialog to the design-working-group for UI/UX feedback, but the gist of it is that you can change the settings and get a live preview of the data (with settings applied).

The ideas I have for an export dialog are follow:
* default settings are fetched from the system settings
* Any setting can easily be overridden when exporting a file
* A checkbox to save the settings that is used in the dialog as system settings.
* Maybe a checkbox to always use the system settings and not show the dialog (not sure yet)

---

All this is done from the perspective of a spreadsheet user, I haven't really used Assembly so I might miss something from that point of view.

In assembly, how is BOM handled, are BOM saved automatically or is it on a tool button click?
If it is automatically, then no dialog should pop up and we need to ensure that the user knows how to configure it.
If manually exported, then I don't see a problem with a popup (or am I wrong here?)

mwganson, 2024-10-08:
Dialog with a preview is a great idea.  An option to skip the dialog in the future would be good, too, but that also would need an easy way to re-enable the dialog, perhaps by adding another option to the preferences.

H-Tsubota, 2024-10-09:
Thank you to everyone who left comments.

For patch2, it seem to that there is no severe problem. So I submitted a pull request for patch2 (#17140).

For patch3, there seem to be some difference in opinion. I personally favor "comma-delimited CSV" default, but I'm not sure I can convince people who favor "tab-delimited CSV" default. So I gave the patch3 up.

@hyarion

In these patches, I focused on minimizing code changes and less side effects. And my approach for this issue is "Done is better than perfect".

I think that introducing Python csv module as CSV import/export engine is great and perfect idea. I have no objection to that at all if you try to implement it.

Also I have no objection to "Exporting dialog" idea. But, in personally, I like a silent, smart GUI. If I have to be forced to push "OK" button for every export, I would be frustrating. So I request a "Don't show this dialog again" checkbox.

In Assembly workbench, BOM file exporting starts from pushing "Export" button. However, I think that the export dialog should probably be implemented in a way that is independent of the calling workbench, since it may be used from another workbench in the future.

If you don't intend to use this issue for your export dialog implementation, please close this issue. Thank you.

hyarion, 2024-10-09:
@H-Tsubota We can keep this issue open. I think it is a great case of the confusion regarding tabs in CSV.

I'll make sure to connect the PR to this issue when it's ready.

closed 2024-10-14 by yorikvanhavre