FreeCAD/FreeCAD #16942: Error: "Creating a popup with parent, QWidgetWindow(0xaaaaf8f5af60,name="Gui::MainWindowClassWindow") which does not match the current topmost grapping popup"

peterwwillis, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello, I just migrated a project from FreeCAD 0.21.2.33771 to the latest beta/RC.

I am also now using a different platform; before I had an X11 Linux system, now I am using a distro with Wayland (sadly...)

In the old system, I could pop out the Model & Report View panels, and move them to a second monitor, so my main monitor just has the 3d view, and my second monitor has the Model/Report panels. Worked great! Tons of space, very productive. When I would reopen FreeCAD it would restore them the right way, fantastic.

Now on this newer system when I try to pop out the panels, they seem to pop out, but I can't drag them with my mouse. I can see for a brief moment a blue window shade in the corner of the main window, then nothing. The panel doesn't go anywhere. I sometimes see this error:

```
01:07:04  (qt.qpa.wayland) eglSwapBuffers failed with 0x300d, surface: 0x0
```

So I decided, let's move the whole FreeCAD window over to the second monitor, then pop the panels out there, then drag the main window back to the first monitor, and see if that works.

Well, it almost works.... but then it throws some Wayland errors on the main window in the Report panel.

Sometimes it will be this error:

```
01:19:49  The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
```

And once, I'm still not sure how (trying to reproduce it - but basically doing the above), it resulted in this error:

```
(qt.qpa.wayland) Creating a popup with a parent, QWidgetWindow(0xaaaaf8f5af60,name="Gui::MainWindowClassWindow") which does not match the current topmost grabbing popup, QWidgetWindow(0xaaab1027d560, name="&ViewWindow") With some shell surface protocols, this is not allowed. The wayland QPA plugin is currently handling it by setting the parent to the topmost grabbing popup. Note, however, that this may cause positioning errors and popups closing unexpectedly. Please fix the transient parent of the popup.
```

I'm attaching some screenshots. Note the second one also has some strange transparency artefacts

[image]
[image]

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38804 (Git)
Build type: Release
Branch: (HEAD detached at b0d847ca15)
Hash: b0d847ca158cd49f4ec1ab806c5e0b618a50d387
Python 3.11.9, Qt 6.7.2, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * A2plus 0.4.67
  * CADExchanger
  * CurvedShapes 1.0.10
  * Curves 0.6.48
  * ExplodedAssembly
  * fasteners 0.5.29
  * frame 0.1.1
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.26
  * SteelColumn
  * woodworking 0.21.2.33771
  * 3DfindIT 1.2.0
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Anything else?

Now for the weird part... this is running on Asahi Linux, on a `Apple MacBook Pro (16-inch, M1 Max, 2021)`. FreeCAD beta of the 1.0.0RC1, as shown in version output (thanks for releasing that btw!).

As for the functionality: it appears that if I'm careful, I can now drop the FreeCAD main window in one monitor screen, pop out the panels, then move the FreeCAD main window back to the other monitor. Then I use my "Windows key+Left mouse" shortcut to manually move the panel windows in the screen, then hover my mouse around the panel corners, and I can resize them. So if I'm very careful, I can get the basic functionality I had before... but it seems to be rather touchy/buggy compared to the old stable release.

Thanks in advance for any help! If you need more debugging output/reports, please link me to some instructions, as I don't know how to provide more debugging info at the moment, but would be happy to.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pinkavaj, 2024-10-02:
Can you reproduce the issue with older version of FreeCad (eg 0.21.2.33771) on the current platform? So we can know if it is by FreeCad or platform change?

peterwwillis, 2024-10-02:
Unfortunately I can't run the older version of FreeCAD on this platform as there is no Flatpak build for it. I tried to build it for this platform but ran into errors (that I filed a bug for)

pinkavaj, 2024-10-02:
The partial transparency issue is described here, if you are able to replicate it (with high reliability) pls consider note it at the ticket https://github.com/FreeCAD/FreeCAD/issues/8341

luzpaz, 2025-07-28:
Is this Flatpak related ?  
* See #21842 

cc @adrianinsaval