FreeCAD/FreeCAD #16944: Wrong tab ordering for Placement dialog

hyarion, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current order is:
... -> X -> Y -> Z -> Apply incremental changes -> Angle

It would feel more natural if it was:
... -> X -> Y -> Z -> Angle -> Apply incremental changes

Screenshot showing the current tab order:
[image]

A side note is that the Rotation dropdown menu cannot be accessed by using tab and should be fixed as well.

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-01:
@PaddleStroke you've created a PR for this dialog, is the tab order changed in it?

hyarion, 2024-10-01:
> @PaddleStroke you've created a PR for this dialog, is the tab order changed in it?

@maxwxyz it is not changed or fixed in that PR. The tab order was discussed there but it was decided to create a new issue for it. This is the new issue.

hyarion, 2024-10-03:
@kriswilk If you are looking for something else to do, maybe this issue would be of interest?

kriswilk, 2024-10-03:
> @kriswilk If you are looking for something else to do, maybe this issue would be of interest?

Sure, I just submitted the fix. For some reason there were custom tabstops set in the dialog. Removing them yields the desired tab order (at least on my Windows 11 PC). Probably just an accident during design of the dialog UI.

PR #16982 (2024-10-03): Placement Dialog: Use default tab order (Fix #16944)
Fixes #16944

Unless the custom tabstops in this UI were deliberate, they should be removed as per this PR. The default order is the desired order.
commit 7e812879 (2024-10-03): Placement Dialog: Use default tab order (Fix #16944)
commit e76ebea7 (2024-10-03): Placement Dialog: Use default tab order (Fix #16944)
closed 2024-10-03 by chennes