FreeCAD/FreeCAD #16945: [Regression] "src/Build/Version.h" make target failed in the first fresh build

ONLYA, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the latest commit in branch "releases/FreeCAD-1-0" ([2b5dca4](https://github.com/FreeCAD/FreeCAD/commit/2b5dca48a50fd8e8cf2531465f40796c1c622ab2)), which is also the latest commit  in the "main" branch ([813ee36](https://github.com/FreeCAD/FreeCAD/commit/813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6)), the first fresh `make` will always fail with the following error:
```
No rule to make target 'src/Build/Version.h', needed by 'src/Build/CMakeFiles/fc_version'. Stop.
```
It will continue to build and finish successfully if I run `make` again.
I suspect it's caused by the changes here: https://github.com/FreeCAD/FreeCAD/commit/49ee02277bc78621514bf1cf578100f94c2a8f04. It will check the `Version.h` dependency before the file is generated, which triggers the error.

The following is my build process for every new commits:
```bash
#!/bin/bash
set -e
FREECAD_BUILD_VER=1.0.0
FREECAD_BUILD_REL=rc2
FREECAD_BUILD_ARCH=amd64

# Remember to clean the build directory
rm -rf build
mkdir build

pushd build

cmake -D BUILD_DRAWING:BOOL=ON -D BUILD_DESIGNER_PLUGIN:BOOL=ON -D BUILD_JTREADER:BOOL=ON -D FMT_DOC:BOOL=ON -D BUILD_FEM_NETGEN:BOOL=ON -D FREECAD_BUILD_DEBIAN:BOOL=ON -D FREECAD_USE_PCL:BOOL=ON -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release ../

make -j$(nproc --ignore=2)

# Build the package and install it
# ........

popd
```
And now I have to make it as:
```bash
#!/bin/bash
FREECAD_BUILD_VER=1.0.0
FREECAD_BUILD_REL=rc2
FREECAD_BUILD_ARCH=amd64

# The same as above

make -j$(nproc --ignore=2)
set -e
make -j$(nproc --ignore=2)

# The same as above

```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38844 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: 2b5dca48a50fd8e8cf2531465f40796c1c622ab2
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.0
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenLight.qss/FreeCAD Light/Fusion
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * A2plus 0.4.67
  * kicadStepUpMod 11.3.9
  * sheetmetal 0.4.26
  * ProDarkThemePreferencePack 1.0.0
  * InventorLoader 1.4.0
  * Render 2024.9.25
  * OpenTheme 2024.9.1
  * ExplodedAssembly
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-01:
Related to 7d0032b8345a71dc54e00f3b863b3f773fbf7da9? @NomAnor

NomAnor, 2024-10-01:
I think make doesn't like the second dependency because its a byproduct.

@ONLYA can you remove `${CMAKE_CURRENT_BINARY_DIR}/Version.h` from the line
https://github.com/FreeCAD/FreeCAD/blob/813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6/src/Build/CMakeLists.txt#L23
and try again?

That seems to work. I make a PR.

ONLYA, 2024-10-01:
> I think make doesn't like the second dependency because its a byproduct.
> 
> @ONLYA can you remove `${CMAKE_CURRENT_BINARY_DIR}/Version.h` from the line
> 
> https://github.com/FreeCAD/FreeCAD/blob/813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6/src/Build/CMakeLists.txt#L23
> 
> and try again?
> 
> If that works, I will make a PR.

Yes. This works :+1:

ONLYA, 2024-10-01:
By the way, running the following `cmake` command would generate the same result as described:
```bash
cmake --build . -j$(nproc --ignore=2)
```
I will check whether that will also fix it.

NomAnor, 2024-10-01:
Thats probably a gamble if the file gets generated before another job needs it. I think removing the extra dependency is the correct fix. 
The fc_version target will always generate a new `Version.h.out` with the command, but because of `${CMAKE_COMMAND} -E copy_if_different` the real `Version.h` is only updated when the contents actually changes.

rdmitry, 2024-10-01:
Also an error in the case of building from tarball, not from repository:

```[   36s] [  1%] Generating version_file
[   36s] cd "/home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/src/Build" && /usr/bin/python3 /home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/src/Tools/SubWCRev.py --srcdir=/home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48 --bindir=/home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/build
[   36s] fatal: not a git repository (or any of the parent directories): .git
[   36s] /bin/sh: line 1: bzr: command not found
[   36s] /bin/sh: line 1: svnversion: command not found
[   36s] /bin/sh: line 1: svn: command not found
[   36s] Unknown version control
[   36s] /home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/build/src/Build/Version.h.out written
[   36s] cd "/home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/src/Build" && /usr/bin/cmake -E copy_if_different /home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/build/src/Build/Version.h.out /home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/build/src/Build/Version.h
[   36s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/FreeCAD-1.0~rc2_g20240930.2b5dca48/build'
[   36s] make[1]: *** [CMakeFiles/Makefile2:3407: src/Build/CMakeFiles/fc_version.dir/all] Error 2
[   36s] make[1]: *** Waiting for unfinished jobs....
[   36s] make[2]: Entering directory '/home/abuild/rpmbui

NomAnor, 2024-10-01:
Does the tarball contain a `Version.h` file? There is a check to not try generate one
https://github.com/FreeCAD/FreeCAD/blob/813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6/src/Build/CMakeLists.txt#L7

If the tarball is missing the `Version.h` file that is different issue. Did you use the "Source code" tarball from the release page?

PR #16952 (2024-10-03): Fix build error with Makefiles for missing Version h
Fixes #16945
closed 2024-10-03 by chennes