FreeCAD/FreeCAD #16947: [Core] Missing add on manager

benj5378, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For some reason, I don't have the add on manager in the expected menu. If I go to Preferences -> General -> Preference Packs, Manage... -> Open Addon Manager... then nothing happens?

[image]

### Full version info

```shell
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38657 +1 (Git)
Build type: Debug
Branch: dpiAboutInfo
Hash: 46d1b6a35f4bff5c3abc993048c3eb7da79014c5
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/159.34
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-01:
```
CommandManager::addTo() try to add an unknown command (Std_AddonMgr) to a widget!
CommandManager::addTo() try to add an unknown command (PartDesign_Sprocket) to a widget!
```

Idk if this is some kind of compile glitch on my device

benj5378, 2024-10-01:
Looks like it's shiboken that can't load multiple workbenches includin AddonManager:
```
During initialization the error "libshiboken2.abi3.so.5.13: cannot open shared object file: No such file or directory" occurred in /home/bensay/Development/FreeCAD-build/Mod/AddonManager/InitGui.py
```

However, the file actually exists??? So why the error???
```sh
$ ls -l /home/bensay/Development/FreeCAD-build/Mod/AddonManager/InitGui.py
-rw-r--r-- 1 bensay bensay 324 Aug 31 15:19 /home/bensay/Development/FreeCAD-build/Mod/AddonManager/InitGui.py
```

benj5378, 2024-10-01:
Seems like it's `import PySide` that fails. `PySide2` is installed, so why is the alias not working?

benj5378, 2024-10-01:
I figured it out. It was `PySide2` not being installed fully. `import PySide2` was fine, but `from PySide2 import QtGui` failed
For some reason, fixing this also fixed an issue with Pivy causing a crash for Coin3D... even though I would think this was totally unrelated

benj5378, 2024-10-02:
Keeping open for other people having the same issue. Already 2 people on Discord today

chennes, 2025-04-24:
Is there a bug to be fixed here?

benj5378, 2025-04-24:
@chennes let's close this. Might be sure to incomplete building or version mismatch