FreeCAD/FreeCAD #16950: Spreadsheet Workbench cell text not visible and colours over-ridden by themes

Footleg, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Text in the spreadsheet cells was almost invisible using the FreeCAD classic theme, unless I changed the foreground colour for the text to something other than black. Changing to light or dark theme then fails to show all cell text colours.

With the Classic theme, start a new project, add a spreadsheet and type into a cell. When you exit the cell the text turns very light grey and is almost invisible against the white background. Select the cell and the text is visible against the highlighted cell background. Set a different foreground colour and the text is displayed using that colour when you exit the cell, unless you choose Black when the problem reoccurs.
Next I changed to the dark theme, and all the cell text colours were reverted to black on white, hiding my colour information.
Changing back to Classic theme the colours are restored in the cells. The text in black foreground cells is rendered in very light grey again. Changing the cell background colour to Black or Red makes it visible, but black text on a white cell background is almost invisible with the classic theme.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.4.56
  * freecad.gears 1.0.0
  * LCInterlocking
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

[Spreadsheet text colours.zip](https://github.com/user-attachments/files/17210285/Spreadsheet.text.colours.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-01:
Related or Dupe? 
* #13862
* #11865

Footleg, 2024-10-01:
I'm not sure either of those is the same issue. Primarily I think the addition of themes may have caused new issues here. The most critical bug is that cell text is almost impossible to read in rc2 with the Classic theme.

luzpaz, 2024-10-04:
@Footleg would you mind adding screenshot(s) of the issue?

SynopseX, 2024-10-07:
I have the same issue:

Cells A1 to A4 are filled with Text "A1", "A2", "A3" and "A4" but only selected cell A2 is visible.
After text color changing of the cell A4 to blue it is visible, but after changing to black is not visible again.

[image]

Footleg, 2024-10-07:
Screenshot of spreadsheet in classic theme colours on Windows 11.
I also checked this on my Windows 10 PC and see the same problem.
Cells A1-A8 contain text here. Only the selected cell has visible text. It is barely visible in the other cells here.
[image]

luzpaz, 2024-10-15:
@SynopseX can you add your full [About](https://wiki.freecad.org/About) info for the version you tested on? TIA

luzpaz, 2024-10-15:
cc @kadet1090 can you take a look at this ?

SynopseX, 2024-10-22:
> @SynopseX can you add your full [About](https://wiki.freecad.org/About) info for the version you tested on? TIA

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Germany (en_DE)
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * CAM
  * Draft
  * Fem
  * Help
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * parts_library
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web

luzpaz, 2024-10-22:
@SynopseX Can you add the full About info for RC2 you tested on? TIA

SynopseX, 2024-10-23:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * btl 0.9.9
  * CAM
  * Draft
  * Fem
  * freecad.gears 1.3.0
  * Help
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web

luzpaz, 2024-11-09:
@mwganson any chance you can take a look ?

mwganson, 2024-11-10:
I cannot reproduce on either of these versions.

OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git)
Build type: Debug
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * BOLTSFC
  * ContextMenuEditor
  * CurvedShapes 1.0.12
  * Curves 0.6.51
  * DynamicData 2.69.0
  * fasteners
  * FCGear
  * freecad.gears 1.3.0
  * gears
  * hfcInventorMentorA
  * InvGears
  * lattice2
  * MeshRemodel 1.10.32
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * ThreadProfile 1.92.0

Sembazuru, 2024-11-17:
This appears to be resolved on my Win11 RC4 installation:

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * Behave-Dark-Colors 0.1.1
  * CubeMenu
  * Curves 0.6.51
  * Defeaturing 1.2.2
  * fasteners 0.5.30
  * fcgear 1.0.0
  * Gridfinity 0.6.0
  * Gridfinity.backup1724615549.0074763 0.5.0 (Disabled)
  * Gridfinity.backup1731267763.7500894 0.5.0 (Disabled)
  * InventorLoader 1.4.0
  * kicadStepUpMod 11.4.1
  * lattice2 1.0.0
  * Lithophane
  * Manipulator 1.5.7
  * ProDarkThemePreferencePack 1.0.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.5.5
  * ThreadProfile 1.92.0

Footleg, 2024-11-19:
I downloaded FreeCAD_1.0.0RC4-conda-Windows-x86_64-py311.7z to retest (on Windows 11) but still see the issue. Text typed into a new spreadsheet in a new document is rendered almost invisible in default theme (white cell background, very light grey text if I set the text colour to Black).

I am on the following:
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.4.56
  * freecad.gears 1.0.0
  * LCInterlocking

luzpaz, 2024-11-20:
See 
* https://github.com/FreeCAD/FreeCAD/issues/17984#issuecomment-2489415997

n8henrie, 2025-09-16:
I have this issue on macos (both macos 15.6 and now macos 26):

```
OS: macOS 26.0
Architecture: arm64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * parts_library
  * 3D_Printing_Tools
  * Help 1.0.3
```

Can confirm that by default I can't see the text in spreadsheet cells unless I highlight the cell. Changing foreground color to blue or green works, when I change it back to black it is essentially invisible again.

Screenshot below I have changed A3 to green, but if you squint (really hard) you can see there is also text in A2, A4, and C2-C4.

[image]

I see in the Python console:

```
App.ActiveDocument.Spreadsheet.setForeground('A2:A2', (0.000000,0.000000,0.000000))
App.ActiveDocument.recompute()
```

Manually using `(1, 1, 1)` turns the text to pure white, manually doing `(1, 1, 1)` goes back to this light gray color.

JohnOCFII, 2025-09-16:
I'm not seeing the same issue.  Can you restart in Safe-Mode to check if you get the same results?   Curious if there is something in your existing preferences/configurations that might be influencing this situation?

[image]

```
OS: macOS 15.7
Architecture: arm64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.8.23
  * manifest.json
  * Telemetry 1.0.2
```

alandsidel, 2025-11-13:
> > It only seems to happen when I select black as the text color, other colors all work as expected.
> 
> Can you provide the detailed information from the "About Box?" I'm not seeing the issue, but note that I'm NOT using the same theme as those reporting the issue. Also - can you re-test in safe mode? That will give another data point as to whether something in your particular configuration is impacting this issue.

Hre you go:

```
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

It does not happen in safe mode, but I do not believe I've made any changes that could affect this either; I wouldn't even know where to look for them.

alandsidel, 2025-11-13:
Additional datapoint:  In safe mode the about box info is different.  Maybe this is the key?

```
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

Stylesheet, theme, and qtstyle are all different in safe mode.

JohnOCFII, 2025-11-14:
> Stylesheet, theme, and qtstyle are all different in safe mode.

```Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default```

Yes, I believe the issue comes up because of the theme.  It may be the theme itself, or the possibility of some older theme-related configuration parameter hanging around from a possible previous configuration.

One thing you could try is to choose a different theme, such as FreeCAD Light or FreeCAD Dark from the theme menu.  Run with that for a few minutes, and if you'd still like to go back to Classic, then re-choose that, and see if it resets the theme properly.

[image]

xtemp09, 2026-10-22:
I can't reproduce the issue. I can say that Dark Theme is really unusable.

<details>

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0RC2.20261018 (Git shallow)
Build date: 2026/10/18 05:49:18
Build type: Release
Branch: (HEAD detached at 9fb16e7f9)
Hash: 9fb16e7f981adb09f7d7f130ce085d68e9825358
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/102.706/1
OpenGL version/vendor/renderer: 4.6.0 - Build 30.0.101.1273/Intel/Intel(R) UHD Graphics 630
```

```ruby
OS: Ubuntu 24.04.3 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 26.3.0RC2.20261018 (Git shallow) AppImage
Build date: 2026/10/18 05:49:18
Build type: Release
Branch: (HEAD detached at 9fb16e7f9)
Hash: 9fb16e7f981adb09f7d7f130ce085d68e9825358
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/96.0525/1
OpenGL version/vendor/renderer: 4.3 (Compatibility Profile) Mesa 25.2.8-0ubuntu0.24.04.2/VMware, Inc./SVGA3D; build: RELEASE;  LLVM;
```

```ruby
OS: Ubuntu 24.04.3 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 27.1.0dev.49291 (Git)
Build date: 2026/10/20 14:26:52
Build type: Debug
Branch: master
Hash: e12d8132665fd1a2fcc5e0a7fa55768b1aeea563
Python 3.12.3, Qt 6.11.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.2.4, Clipper2 2.0.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/96.0525/1
OpenGL version/vendor/renderer: 4.3 (Compatibility Profile) Mesa 25.2.8-0ubuntu0.24.04.2/VMware, Inc./SVGA3D; build: RELEASE;  LLVM;
```

</details>

closed 2026-11-10 by maxwxyz