FreeCAD/FreeCAD #16951: Rotating a datumPlane does not update rotation of sketch until sketch is opened and closed

glasscanvas, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Sketched a circle on xy plane, radius 50mm
2. Created datumPlane on XY plane with a Z offset of 50mm.
3. Sketched a circle on xy datumPlane
4. Created a datumPlane001 on XY plane with a Z offset of 100mm.
5. Sketched a rectangle on offset datumPlane001, 20x20mm
6. selected the 3 sketches and performed additive loft
7. All OK
8. Now rotate datumPlane001 by 45 degrees and close
9. rectangle does to follow rotation until
10. the sketch of the rectangle is opened and then closed
11. now it refreshes 

Rectangle rotation and position should have updated when the  datumPlane001 plane rotation and/or position are changed and OK is pressed. It requires opening and closing the corresponding sketch to see update.

Thank you very much.

### Full version info

```shell
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Anything else?

[BugReport_20241001.zip](https://github.com/user-attachments/files/17212411/BugReport_20241001.zip)

Project file and 
before screenshot with DatumPlane001 position and rotation returned to 0,0,0, note that Sketch002 is still in the previous rotation and position

after screen shot shows Sketch002 returned to correct position but only after opening and closing the sketch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-01:
Is this also an issue in the stable 0.21.2 version?

FlachyJoe, 2024-10-01:
I can't reproduce on 

> OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
> Word size of FreeCAD: 64-bit
> Version: 1.1.0dev.38843 (Git)
> Build type: Debug
> Branch: dev
> Hash: 77eaa363e10354e62cf90c4c95d5c8a586843098
> Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
> Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]

Refresh occurs when property is validated.

@glasscanvas 
> rotate datumPlane001 by 45 degrees **and close**

What do you close here?

mwganson, 2024-10-01:
I cannot reproduce here.  Sketch instantly updates upon rotating datum plane 001 using the provided file.  It could be something with Mac builds only.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default

glasscanvas, 2024-10-01:
This issue is on an M2 Mac Airbook. I can try this on a PC but not until next week. 
What I close in step 8 is editing of datumPlane001 (name in attached file)

mwganson, 2024-10-01:
In one of your images it clearly shows that the datum plane needs to be recomputed.  Compare also the refresh icon on the toolbar in the 2 images, it's the one between the scissors and the right curved arrow for the redo command.

I would not consider this a bug.  It is common that you will need to recompute after some changes.

glasscanvas, 2024-10-02:
Thank you very much for your help. 
Question: Can refresh be set automatically?
(it is probably best to refresh manually)

mwganson, 2024-10-02:
I don't know of an option to automatiically recompute.  I just monitor the icons and recompute manually as necessary.

Syres916, 2024-10-05:
Personally this is one of the few situations where I would add the recompute https://github.com/Syres916/FreeCAD/commit/ddbaa1a94f470016ac0b002cb73399778b46c2df especially if you look in the Python Console, just before the `resetedit()` there is a `recompute()` so that tells me the developer intended that the model be updated IMHO.

Syres916, 2024-10-09:
@glasscanvas could you please either close this issue as _education_ or advise if you believe the _Recompute_ should be added, TIA.

closed 2025-08-13 by github-actions[bot]