FreeCAD/FreeCAD #16953: Sketcher: Double click to select connected wires not working

maxwxyz, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Double click to select connected wires (https://github.com/FreeCAD/FreeCAD/pull/11925#issuecomment-2384948891) not working anymore.
It was also reported by @pierreporte here https://github.com/FreeCAD/FreeCAD/issues/16198 but only for slots, but the feature is not working at all.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-10-08:
`ViewProviderSketch::isSelected` is failing (always return false)
because `Gui::Selection().isSelected` is failing (always return false)
can be tested in python : `Gui.Selection.isSelected(obj, "Edge3")`

Dunno why yet

PaddleStroke, 2024-10-08:
`Gui.Selection.isSelected(obj, "Edge3")`
fails if in edit mode
works if not in edit mode

PaddleStroke, 2024-10-08:
In edit mode : 
```
>>> selection = Gui.Selection.getSelectionEx("*", 0)
>>> selection[0].FullName
"(App.getDocument('offset_tests').getObject('Sketch002'),[';g95.edge3',])"
```
Outside edit mode : 
```
>>> selection = Gui.Selection.getSelectionEx("*", 0)
>>> selection[0].FullName
"(App.getDocument('offset_tests').getObject('Sketch002'),[';g95;SKT.Edge3',])"
```

Hmm it seems the subname is not exactly the same while in edit mode.

PaddleStroke, 2024-10-08:
@chennes does that ring a bell to you?
`isSelected `is also used by `preselectToSelection` so if it is failing we should have weird behaviors in selection of geos

PaddleStroke, 2024-10-08:
Indeed we do have a bug in selection actually. If you select a geo in sketcher, then click it again, then it does not unselect as it should. Because of the same reason, isSelected fails.
@maxwxyz do we have an issue about this already?

PaddleStroke, 2024-10-08:
So I think it has something to do with the not capital 'E' of 'edge3'.
Searching the codebase for "edge" with match case, we find in SketchObject several references to this. So it seems that it is done on purpose?

And indeed if I change to lower case in the wire selection code then it works.

`    const char* type1 = (clickedGeoId >= 0) ? "edge" : "externalEdge";`
instead of 
`    const char* type1 = (clickedGeoId >= 0) ? "Edge" : "ExternalEdge";`

The changes that introduced "edge" in SketchObject is 3 months old, commits by RealThunder. I'm going to check Linkstage3 for more information

PaddleStroke, 2024-10-08:
That's it I found it.
 + getSketchObject()->convertSubName(subNameSuffix) needs to be added to `ViewProviderSketch::isSelected` & `ViewProviderSketch::rmvSelection`

maxwxyz, 2024-12-20:
@PaddleStroke it works now, but not for defining external geometries.