FreeCAD/FreeCAD #16954: [TechDraw] Make dimension coordinates relative to dimension lines

benj5378, 2024-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The position coordinates should be relative to the dimension lines and not the text.

This would also make the dimension position independent of label side (right or left to the dim line), independent to font size (that changes the location of the base font line), and would make coordinate handling much simpler when introducing more advanced labels. Unsure, but perhaps also when adding a second line by dimension tolerancing?

Dimension text should be placed according to the base font line. This will fix the different label gaps depending on the decimal seperator.

For instance when [#13617](https://github.com/FreeCAD/FreeCAD/issues/13617) is implemeted, adding a labeling symbol will change the dimension line location (due to the new bounding rect), when it's actually the label location that should be changed. So this would call for either overly complicated coordinate compensation and maths to fix - or just change XY to be relative to the dimension line (instead of label).

As per https://forum.freecad.org/viewtopic.php?t=90323

### Full version info

```shell
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Debug
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-10-02:
see also https://forum.freecad.org/viewtopic.php?style=5&p=672901#p672901

WandererFan, 2024-10-19:
Some other issues re dimension text positioning:
TechDraw: Default position + snapping of dimensions #13832
TechDraw: Snap and Create Dimensions with Distance to Model Lines #11636
TechDraw: Snapping Dimensions: Angles/Diameter/Radii #13831

luzpaz, 2024-10-19:
@WandererFan An aside, you can save yourself cut/paste time by putting the github issue in a markdown list and github automagically renders the ticket name for you:
```
* https://github.com/FreeCAD/FreeCAD/issues/13832
* https://github.com/FreeCAD/FreeCAD/issues/11636
* https://github.com/FreeCAD/FreeCAD/issues/13831
```
Produces: 
* https://github.com/FreeCAD/FreeCAD/issues/13832
* https://github.com/FreeCAD/FreeCAD/issues/11636
* https://github.com/FreeCAD/FreeCAD/issues/13831

WandererFan, 2024-12-28:
`Dimension text should be placed according to the base font line. This will fix the different label gaps depending on the decimal seperator.`

This is appears to be fixed by  PR #18771.

luzpaz, 2024-12-28:
@WandererFan do you want to close this ticket, ITC ?

WandererFan, 2024-12-29:
No. PR #18771 only solves the issue with aligning text with descenders.  Changing how dimension position is calculated is still valid.