FreeCAD/FreeCAD #16955: Manual update dialog for editable field doesn't update

j8spencer, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

My model was originally named "BaseSideWall", which is also the name of the first body in the model. I had a couple of tech drawings that show the model name as the title. I updated the model name to "Tack Trunk", and did TechDraw | page | update editable field. But this title update was not listed as one of the changes. I cancelled, and then just clicked on the title field in the drawing page, and I got the image below, where I clicked "auto-update." On hitting OK, the drawing title changed to "Tack Trunk," but this was not listed in the editable field update dialog as the value that would go in on OK.
[image]

I've also noticed that other fields have difficulty updating on pages. Sheet comes out as 1/1 on the first page, but the second page gets "2/2" until I update it, then it goes to "1/1." I could not get the first page to change to "1/2."

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * A2plus 0.4.67
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-03:
Could you upload the example file as .zip and maybe share the template which was used.

j8spencer, 2024-10-03:
Here you go. I built the template by copying one of the std ones and removing stuff I don't want, so it deserves some skepticism, but it mostly works...

[trunk.zip](https://github.com/user-attachments/files/17250329/trunk.zip)

maxwxyz, 2024-10-04:
I cannot reproduce on my side. Both work as expected, page is 1/2 and 2/2. Title changes correctly.
@WandererFan are you on Linux and can reproduce?

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

j8spencer, 2024-10-04:
I just noticed today that if I click on the Sheets field and choose "autofill" check box, the text in the dialog is wrong (since I've added new pages), but on close, the drawing page _does_ update correctly. I didn't know I had to choose "autofill" before, maybe that confused me. Is the manual update dialog box expected to show the right autofill values before you hit OK?

WandererFan, 2024-10-13:
The total page count on existing pages is not changed until an autofill operation is triggered on that page.   The existing pages don't know anything about the creation of a new page.   

I'm told that being able to assign page numbers manually is important since not all pages in a real document are created by FreeCAD.  So we might actually want to number our FreeCAD pages 3, 5, 8, ...

This of course makes auto numbering problematic.

WandererFan, 2024-10-13:
Other than the total page count on existing pages, this appears to work fine on linux.

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 +1 (Git)
Build type: debug
Branch: fixDimAutocorrect17189
Hash: fcd60659ab43e9369a4be46e086b29e162571266
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

closed 2024-12-28 by WandererFan