FreeCAD/FreeCAD #16957: [TechDraw] Dimensions in view don't match part

j8spencer, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I created a view on a page with, I think, a combination of "camera view" and the "Top" arrow. I scaled the view as 1:3, and I also have a default dimension format string of "%.3w". When I add dimensions to the drawing, what should be measurements of 40.0, 41.5, and 45.0 deg. come out as 39.996, 41.499, and 45.006 deg.

Here are the view properties and the result.

[image]

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * A2plus 0.4.67
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-10-02:
Look at your projection direction.  It is not standard Z (0,0,1), but slightly off at (0.01, -0.01, 1).   This will cause the unexpected dimensions.

luzpaz, 2024-10-02:
Closing as PEBCAK

Note: please update to RC2. TIA

closed 2024-10-02 by luzpaz