FreeCAD/FreeCAD #16958: Recomputation required dialog in some files opens every time document is loaded

hyarion, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Current behavior:
The attached file shows the dialog "Recomputation required" when loaded, even after a recompute + save.

Preferred behavior:
The dialog shouldn't need to be shown if the document has been recomputed and saved with the same version.

---

From what I can see the dialog is shown because the `Part::PropertyPartShape::Restore` functions encounters `ElementMap` attributes with value `?` in some of the `removalshape` properties.

To find these in `Document.xml`, search for `ElementMap="?"` and you will get 6 results.

A breakpoint can be set on [this line](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Part/App/PropertyTopoShape.cpp#L369) to see where it sets the flag `Status::RecomputeOnRestore` to `true`.

I've marked this as CAM related as the issues the attached file is limited to CAM operations.

---

This issue was reported from Knox Machining on discord and he sent this file where the issue can be reproduced:
[spoilboard6-20.10-dog-holes.FCStd.zip](https://github.com/user-attachments/files/17220762/spoilboard6-20.10-dog-holes.FCStd.zip)

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6

and

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-02:
I've also noticed this issue in the RC with files created in 0.22.dev PartDesign. The recompute sometimes break the model too.
@chennes any idea?

hyarion, 2024-11-02:
Cannot reproduce this anymore, closing issue

forest-hermit, 2024-11-22:
It is happening to me with the 1.0 stable.

willis936, 2025-01-12:
> It is happening to me with the 1.0 stable. Could we reopen it?

I am also having this issue.

maxwxyz, 2025-01-12:
Please add an example file created with 1.0.0

willis936, 2025-01-12:
> Please add an example file created with 1.0.0

[House.zip](https://github.com/user-attachments/files/18389889/House.zip)

hyarion, 2025-01-12:
> > Please add an example file created with 1.0.0
> 
> [House.zip](https://github.com/user-attachments/files/18389889/House.zip)

I cannot reproduce the issue with the file in 1.0.0, it loads without problem for me.

```
OS: macOS 14.5
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.6.11
```

maxwxyz, 2025-01-12:
@willis936 can you try restarting FreeCAD in safe mode (Help>restart in safe mode) and then open this file?

willis936, 2025-01-12:
I am able to recreate this issue with the uploaded file even when restarted in safe mode.

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

hyarion, 2025-01-12:
I tried in safe mode on 1.0.0 (full version above) and still no recompute dialog for me 🤔 
Can you download the file you uploaded to see that it isn't another file you are opening?

NomAnor, 2025-01-12:
I also have a file with this problem. Looking in the `Document.xml` it's because of some `Mirror` and `LinearPattern` inside `MultiTransforms`

```
<Object name="LinearPattern002" Extensions="True">
<Extensions Count="1">
    <Extension type="App::SuppressibleExtension" name="SuppressibleExtension">
    </Extension>
</Extensions>
<Properties Count="19" TransientCount="1">
    <_Property name="_Body" type="App::PropertyLinkHidden" status="251658240"/>
    <Property name="BaseFeature" type="App::PropertyLink" status="8">
        <Link value=""/>
    </Property>
    <Property name="Direction" type="App::PropertyLinkSub">
        <LinkSub value="Y_Axis001" count="1">
            <Sub value=""/>
        </LinkSub>
    </Property>
    <Property name="ExpressionEngine" type="App::PropertyExpressionEngine" status="67108864">
        <ExpressionEngine count="2">
            <Expression path="Occurrences" expression="App__VarSet001.BoxDepth"/>
            <Expression path="Offset" expression="&lt;&lt;DesignParameters&gt;&gt;.CellDepth"/>
        </ExpressionEngine>
    </Property>
    <Property name="Label" type="App::PropertyString" status="134217728">
        <String value="LinearPattern002"/>
    </Property>
    <Property name="Label2" type="App::PropertyString" status="67108992">
        <String value=""/>
    </Property>
    <Property name="Length" type="App::PropertyLength" status="4">
        <Float value="0.0000000000000000"/>
    </Property>
    <Property name="Mode" type="App::PropertyEnumeration">
        <Integer value="1"/>
    </Property>
    <Property name="Occurrences" type="App::PropertyIntegerConstraint">
        <Integer value="1"/>
    </Property>
    <Property name="Offset" type="App::PropertyLength">
        <Float value="25.0000000000000000"/>
    </Property>
    <Property name="Originals" type="App::PropertyLinkList">
        <LinkList count="0">
        </LinkList>
    </Property>
    <Property name="Placement" type="App::PropertyPlacement" status="8388620">
        <PropertyPlacement Px="0.0000000000000000" Py="0.0000000000000000" Pz="0.0000000000000000" Q0="0.0000000000000000" Q1="0.0000000000000000" Q2="0.0000000000000000" Q3="1.0000000000000000" A="0.0000000000000000" Ox="0.0000000000000000" Oy="0.0000000000000000" Oz="1.0000000000000000"/>
    </Property>
    <Property name="Refine" type="App::PropertyBool">
        <Bool value="false"/>
    </Property>
    <Property name="Reversed" type="App::PropertyBool">
        <Bool value="true"/>
    </Property>
    <Property name="Shape" type="Part::PropertyPartShape">
        <Part ElementMap="?" file="LinearPattern002.Shape.brp"/>
        <ElementMap/>
    </Property>
    <Property name="ShapeMaterial" type="Materials::PropertyMaterial">
        <PropertyMaterial uuid="7f9fd73b-50c9-41d8-b7b2-575a030c1eeb"/>
    </Property>
    <Property name="Suppressed" type="App::PropertyBool" status="8">
        <Bool value="false"/>
    </Property>
    <Property name="SuppressedShape" type="Part::PropertyPartShape">
        <Part ElementMap="0.4" file="LinearPattern002.SuppressedShape.brp"/>
        <ElementMap/>
    </Property>
    <Property name="TransformMode" type="App::PropertyEnumeration">
        <Integer value="0"/>
    </Property>
    <Property name="Visibility" type="App::PropertyBool" status="648">
        <Bool value="false"/>
    </Property>
</Properties>
```

I can't provide the file, but I can help with the debugging.

```
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +469 (Git) (Tag v1.0.0 with a cherry pick to get it to compile)
Build type: Release
Branch: (HEAD losgelöst von 1.0.0)
Hash: 1a03c20ebe3df99691542398fb07d5171cd0d9da
Python 3.13.1, Qt 6.8.1, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * ThreadProfile 1.84.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.3.0
  * fasteners 0.5.27
```

The House file from above shows the dialog again even after a recompute and save.

willis936, 2025-01-12:
> I tried in safe mode on 1.0.0 (full version above) and still no recompute dialog for me 🤔 Can you download the file you uploaded to see that it isn't another file you are opening?

In the test I did for my last comment I downloaded the file I uploaded to the previous comment and tried loading it fresh.

I tested again just now.  I did notice that it doesn't trigger recompute if it's the first file opened on FreeCAD launch, but if you close it then open it again it triggers recompute.

hyarion, 2025-01-12:
Closing document (without saving) and reopening it again does indeed open the dialog.

NomAnor, 2025-01-14:
For my case with the `MultiTransform` children, I think the problem is because:

1.  https://github.com/FreeCAD/FreeCAD/blob/0b1d7349b0aa95447032d91c946f8187cd5a9b64/src/Mod/Part/App/PropertyTopoShape.cpp#L324-L327 sees a "?" and keeps that in `_Ver`
2. https://github.com/FreeCAD/FreeCAD/blob/0b1d7349b0aa95447032d91c946f8187cd5a9b64/src/Mod/Part/App/PropertyTopoShape.cpp#L95 is never called to clear `_Ver` because `MultiTransform` children don't do anything in `execute()` and don't have a shape (despite having the property). Ideally they should not inherit `Part::Feature`.
3. https://github.com/FreeCAD/FreeCAD/blob/0b1d7349b0aa95447032d91c946f8187cd5a9b64/src/Mod/Part/App/PropertyTopoShape.cpp#L274-L278 stores "?" back into the file, triggering the dialog again on the next load.

I "fixed" it by rewriting it by explicitly checking for "?" so the invalid `_Ver` value is not used:
```c++
auto ver_valid = _Ver.size() && (_Ver != "?");
if(owner) {
    if(!owner->isExporting())
        version = ver_valid?_Ver:owner->getElementMapVersion(this);
}else
    version = ver_valid?_Ver:_Shape.getElementMapVersion();
```

Not sure if this is a good fix, someone with knowledge about the element maps should look over it.

Haven't looked if something similar is happening with the House file.

NomAnor, 2025-01-14:
The House file is fixed with my patch, I must have used the wrong version.

I submited a PR with my patch.

ForrestErickson, 2025-01-21:
Regarding, 

> [maxwxyz](https://github.com/maxwxyz)
> [last week](https://github.com/FreeCAD/FreeCAD/issues/16958#issuecomment-2585766385)
> Collaborator
> Please add an example file created with 1.0.0
> 

Why did you ask for a file created with 1.0.0?  Is this not an issue related to a file created with some version before 1.0.0?  Is that not what migration would be all about or am I missing something?

My case: I have a file first created in 0.19 some time ago, and then edited with 0.21 and now I am trying to migrate to 1.0 with out success.  My file uses Multitransform.

ForrestErickson, 2025-01-21:
## Making the problem reproducible in a simple file.

Making a file in 0.21 with Multitransform

[image]

Open in FreeCAD 1.0

[image]
Press Yes then save.

Then open again and the problem remains

[image]

Here is a zip file with the before and after files:

[Ver0.21MultiTransform.zip](https://github.com/user-attachments/files/18489654/Ver0.21MultiTransform.zip)

chennes, 2025-02-03:
Can we try to reproduce in LinkStage3 and see if that fork also has the problem? Right now our code matches the LS3.

chennes, 2025-02-17:
ACtually, @ForrestErickson can you try with a new weely and see if this now resolved?

closed 2025-02-24 by chennes

ForrestErickson, 2025-08-27:
Regarding, " new weely and see if this now resolved?"
Did anyone try my file, [Ver0.21MultiTransform.zip](https://github.com/user-attachments/files/18489654/Ver0.21MultiTransform.zip) to see if they can first reproduce the problem and then test of the " new weely" fixes it?

And, what EXACT version.  New is not a version.  It is now 20250827 as I type.

PR #19068 (2025-02-24): Always update the element map version if it's not valid or unknown
`_Ver` can contain "?" after loading and in some circumstances this is not cleared (often by `setValue()`) before saving (`MultiPattern` children or empty `Part::Feature`). When saved the same "?" is then written again to the file causing the recomputing dialog to pop up again and again.

This not only checks for an empty `_Ver` but also explicitly for "?" and then regenerating the version if necessary.

Fixes #16958 for the provided test file and the mentioned MultiPattern case.
This should be reviewed by someone with knowledg [... cut]