FreeCAD/FreeCAD #16959: Sheet Metal Workbench 0.4.26 causes problems with previously working projects

salvatore28, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sheet Metal Workbench 0.4.26 (FreeCAD 0.21.2) causes problems with previously working projects (using Sheet Metal Workbench 0.4.24 and FreeCAD 0.21.2), see image:

[image]

Example project that works without errors with version 0.4.24:

[Modulo principale - Pannello anteriore.zip](https://github.com/user-attachments/files/17221074/Modulo.principale.-.Pannello.anteriore.zip)

Steps to reproduce the problem:

1. click on Fold
2. set unfold to true and update the view
3. set unfold back to false and update the view

Alternative steps to reproduce the problem:

1. right click on Fold
2. click "Recompute object"

Temporary workaround to resolve the problem:

Revert to Sheet Metal Workbench version 0.4.24

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * kicadStepUpMod 11.3.9
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.24
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Problem already discussed here:

[https://forum.freecad.org/viewtopic.php?p=785079](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-10-02:
@shaise is it possible that this can be moved over to https://github.com/shaise/FreeCAD_SheetMetal/Issues ? I believe https://github.com/shaise/FreeCAD_SheetMetal/pull/368 may well be responsible for the regression

shaise, 2024-10-02:
@Syres916 , Definitely.

luzpaz, 2024-10-02:
Closing as more relevant to SheetMetal Addon.

closed 2024-10-02 by luzpaz

salvatore28, 2024-10-04:
> @shaise is it possible that this can be moved over to https://github.com/shaise/FreeCAD_SheetMetal/issues ? I believe [shaise/FreeCAD_SheetMetal#368](https://github.com/shaise/FreeCAD_SheetMetal/pull/368) may well be responsible for the regression

hello,

should I open a new issue on ...FreeCAD_SheetMetal/issues or will this be moved automatically?

shaise, 2024-10-04:
yes, please reopen it on [ SheetMetal/issues]( https://github.com/shaise/FreeCAD_SheetMetal/issues )
and reattach the example file