FreeCAD/FreeCAD #16961: Quick Measure broken

maxwxyz, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Quick measure only works in bodies, not imported STEP files or Solids.

Steps to reproduce:
1. Create a cube in Part Design
2. Select an edge --> quick measure works
3. Switch to Part WB and create a solid simple copy of the body
4. Select an edge --> quick measure does not display anything

[image]

[image]

@PaddleStroke FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-10-02:
You say broken, was this working before?
I can confirm that Part/Create simple copy objects are not working. Which is quite surprising since they seems to be regular Part::Feature objects.

PaddleStroke, 2024-10-02:
Oh ok I know why it is not working... It is a bug due to the (bad) way we are currently filtering the objects for quickmeasure : 
```
    auto shouldSkip = [](App::DocumentObject* obj) {
        std::string vpType = obj->getViewProviderName();
        auto* vp = Gui::Application::Instance->getViewProvider(obj);
        return (vpType == "SketcherGui::ViewProviderSketch" && vp->isEditing())
            || vpType.find("Gui::ViewProviderOrigin") != std::string::npos
            || vpType.find("Gui::ViewProviderPart") != std::string::npos
            || vpType.find("SpreadsheetGui") != std::string::npos
            || vpType.find("TechDrawGui") != std::string::npos;
    };
```

So what happens is that simple copies are basic Part::Feature object. And so it's viewprovider is the basic one for those : `PartGui::ViewProviderPart`. And this is triggering `vpType.find("Gui::ViewProviderPart") != std::string::npos` even though this was intended to filter out App::Part.

Anyhow this is going to be fixed by https://github.com/FreeCAD/FreeCAD/pull/16921 since it's scrapping this bad `shouldSkip`.

maxwxyz, 2024-10-02:
It also does not work for imported STEP files, which was working before. So it's a regression IMO

PaddleStroke, 2024-10-02:
Yes it used to work actually and is a regression from https://github.com/FreeCAD/FreeCAD/commit/cf2b98b519c903a3c46be9d53c58f0e56dbfc7e2

I was wondering why I'd have made this filtering out based on the VP types in the first place. But I actually didn't so that explains that. Initially I made a filtering on VP types because I wanted to exclude in-edit sketches, so I needed vp->isEditing(). Then Syres built on top on that to add more types to exclude which caused the regression.

Anyway as said it's fixed by #16921

closed 2024-10-03 by chennes