FreeCAD/FreeCAD #16962: Regression: older, working 3Dconnexion devices (spacemouse) dropped in Windows

dewi-ny-je, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The PR https://github.com/FreeCAD/FreeCAD/pull/12929 introduces support for NavLib, the 3Dconnexion proprietary driver interface, but by doing so it drops native support for older and yet perfectly working devices, the support for which has been discontinued in 3Dconnexion NavLib drivers.

Since the old native interface can still be enabled at compile-time https://wiki.freecad.org/3Dconnexion_input_devices#Windows maybe there's a way to compile both and have users pick what they need, or automatically picking based on some device ID.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.47
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GAZ082, 2024-10-02:
A solution would be allowing us to select the driver for the device.

luzpaz, 2024-10-02:
cc @pskowronskiTDx

luzpaz, 2024-10-20:
Anyone want to take point on this ?

cculpepper, 2024-10-21:
It seems we might just be able to add back the menu by reverting most of this commit (making sure to keep the typo fix for the CMAKE options) https://github.com/FreeCAD/FreeCAD/pull/14589

I'll see if I can get a build environment up and running.

dewi-ny-je, 2024-11-11:
While the change of approach can be seen as regression, it's also true that there are now good open source projects to build a spacemouse-compatible device for very little money:
https://github.com/AndunHH/spacemouse

Home-users could build one of those for 20-25 Euro or get a 2nd hand spacemouse for about 70-90 Euro. At this low price it's even difficult to justify the time to recompile FreeCAD.

cculpepper, 2024-12-21:
Initial testing doesn't work, Pulled down the latest windows nightly build, no luck with the mouse.

Pulled open a 0.20 version and the mouse worked great there. I was not able to find the same Tools > Customize menu for the spaceball in the nightly version. Seeing if I need to enable something now.