FreeCAD/FreeCAD #16965: Resize a STL using resize in draft create a blank clone

eagabriel, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Import a STL, go to draft, choose resize, resize it like 1,1 the STL will be resize at least as a preview, is it possible to see it resized, but when you apply a blank clone is created.
Using FreeCAD_weekly-builds-38843-conda-Windows-x86_64-py311
The same works in 0.21

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.15
  * fasteners 0.5.24
  * Render 2024.8.4
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-02:
You cannot scale a mesh with Draft_Scale. To my knowledge this has never worked. It is of course confusing that the preview suggests differently.

Try creating a Link instead. You can change the "Scale Vector" property of a Link. This property is not visible by default. You have to right click in the Property editor and select "Show all"/"Show hidden" from the context menu.

eagabriel, 2024-10-02:
It works perfectly in 0.21 and is what I'm doing whan I need it.

Even in must recent weekly review It works, showing the resized stl over the original one, but whan click ok it generate a blank clone.

I didn't find this link option, could you give me further detail?

Roy-043, 2024-10-02:
Can you post a file with an example created in V0.21?

Documentation about Links:
https://wiki.freecad.org/Std_LinkMake

mwganson, 2024-10-02:
Draft Scale was not designed to work with meshes.  Scaling of mesh objects is supported in the OpenSCAD workbench, but this requires installation of the OpenSCAD binary.  Mesh workbench also has a scale mesh feature, but it only supports uniform scaling where the one in OpenSCAD allows for non-uniform scaling.

I was not aware Links could scale meshes, so that is good to know.

eagabriel, 2024-10-02:
Right, It worked with the link process. tks
And if it is not the ideia having a scale function in draft with meshes its ok. 
Perhaps disable the options in draft when a mesh is selected could help.

closed 2024-10-02 by eagabriel