FreeCAD/FreeCAD #16966: Suggestion: Enhance axis display in Sketch Creation.

leoheck, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a new sketch, planes and axes appear on the screen to guide the sketch creation. They can also be clicked to select the sketch plane.

While lines may be too thin to select, axis names serve as a good reference for orienting the user and quickly selecting the plane.

_So, it would be helpful to resize the axis planes to fit the screen when the object is bigger than what’s visible._

This is what we see when the objects fit the screen...
[image]

However, when the objects do not fit the screen... we cannot see those axes names.
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-10-02:
When #16675 will be merged, planes are going to be displayed at a constant screen size independent from zooming level. Though names have been removed.

leoheck, 2024-10-02:
> Names have been removed 

Hmm, interesting. How this helps users to identify (and also select) planes? Without those names the axes won't give extra information when adding a Sketch when all of them are visible at the same time.

pierreporte, 2024-10-02:
How? Well, it doesn’t. From experience it’s not a big deal most of the time, but I agree that there should be a way to show the name without having a big text as today. Maybe show it only at hover?

@PaddleStroke do you think you can use the new LCS axs label font size to display datum names?

leoheck, 2024-10-03:
right now, having them big is what helps selecting the right plan quickly... because the labels have a beefy area to click... It is also possible to click on the thin lines/borders but the user has to target a specific thin place which requires precision.. and this gets worse when using the track pad.

PaddleStroke, 2024-10-03:
In my current PR, the planes face can be clicked, not only the lines. It's possible since they are not too big and they resize automatically. There are no names for the planes, but there are colors and the axis still have their names (see the screenshots in 16675).
So overall I think it's probably an acceptable solution imo.

PaddleStroke, 2024-10-03:
@leoheck perhaps you can compile 16675 to test it out?

closed 2024-12-06 by chennes