FreeCAD/FreeCAD #16968: PartDesign: Thickness doesn't work

Pendelf, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Can't make the thickness 

https://github.com/user-attachments/assets/27a20ecb-fbdd-46ef-81e2-9f00cdf67434

[thickness_1.zip](https://github.com/user-attachments/files/17231593/thickness_1.zip)

It doesn't work in Part either

https://github.com/user-attachments/assets/45558415-664c-4464-814a-a9cc6c071315

In LinkBranch, it works at least inward

https://github.com/user-attachments/assets/a22c72aa-f03d-492a-841d-57959d649ae4

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.29
  * FEMbyGEN 2.2.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.8.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.26
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-02:
Did you check the other Thickness bugs first ? https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+label%3A%22Topic%3A+Thickness%22

Pendelf, 2024-10-02:
https://github.com/FreeCAD/FreeCAD/issues/16206

It might be similar to this, but it's not certain

FEA-eng, 2024-10-02:
There was a similar case on the forum recently: https://forum.freecad.org/viewtopic.php?t=90689

SoulWager, 2024-10-02:
Can you replicate it on the current release candidate or dev version? 

[image]

Pendelf, 2024-10-03:
@SoulWager 
Try using Taper when you extrude a sketch

Pendelf, 2024-10-03:
RC2

https://github.com/user-attachments/assets/b64f92dd-7291-4adb-a4e1-03266a59d03f

SoulWager, 2024-10-05:
Sorry, thought you were using perspective view instead of having a tapered pad. 

There appear to be two issues with the tapered version, one is that the bottom fillet fails to offset if it isn't g2, and the second is that even a shape that will offset(with the G2 fillet) isn't working with thickness. 

Workaround: 
pad rounded rectangle with taper
use subtractive pipe to make a G2 continuous fillet  
make a simple copy(part workbench)  
Downgrade(draft workbench) 
delete top face
upgrade faces into shell
3d offset shell(part workbench)

If the corners need constant radius, rather than being larger at the top than the bottom, you can use blendsurface to make one of the corner fillets with g2 continuity, then mirror it to all four corners and fill in between with ruled surfaces.
[tapered fillet offset.zip](https://github.com/user-attachments/files/17265024/tapered.fillet.offset.zip)

codeotaku, 2026-08-24:
So. this doesn't appear to be a thickness bug..., but rather a bug in the fillet builder where there is a condition that is overzealous in preventing a seam correction (knot reduction) operation from running.  

During fillet construction multiple guide edges are concatenated into a non-periodic B-spline, and for a closed spine, the SetPeriodic() function merges the two endpoint knots into the periodic seam..., but..... the existing code only reduced that seam knot when one of the original edge curves had been approximated by C2.  And this could creates a condition where the seam could have excessive [knot municipality](https://computergraphics.stackexchange.com/questions/358/nurbs-knot-multiplicity), which could result in a discontinuity (that prevents some downstream operations, like thickness, from working). 

I'm going to submit a PR to OCCT.  It seems like the check (in OCCT) was introduced as a fix to some other edge case, but had an unintended side effect that kind of breaks later operations like thickness etc.  

[This is the patch](https://github.com/codeotaku/OCCT/tree/fix-periodic-fillet-c2), if anyone is curious or wants to try it out in the interim. 

[image]

luzpaz, 2026-08-30:
> I'm going to submit a https://github.com/Open-Cascade-SAS/OCCT/pull/1501. It seems like the check (in OCCT) was introduced as a fix to some other edge case, but had an unintended side effect that kind of breaks later operations like thickness etc.

Legend!

Edit:

> So. this doesn't appear to be a thickness bug..., but rather a bug in the fillet builder where there is a condition that is overzealous in preventing a seam correction (knot reduction) operation from running.

Lets remove the labels and the parent issue + rename the ticket, in that case ?

luzpaz, 2026-09-05:
Upstream merged fix https://github.com/Open-Cascade-SAS/OCCT/pull/1501 !

luzpaz, 2026-10-30:
Anyone able to test this on latest OCC version ?

Pendelf, 2026-10-30:
@luzpaz Are we talking about FreeCAD version 27.1?

luzpaz, 2026-11-05:
@Pendelf talking about OCC v8.1.0 IR branch

flaviut, 2026-11-06:
I tested it with FreeCAD on the latest OCCT and its fixed

Seems to be slated for OCCT 8.0.2, perhaps FC 27.1?

[image]

[automated reproducer](https://github.com/user-attachments/files/34487223/issue-16968-macro.zip)