FreeCAD/FreeCAD #16970: BIM: Error when opening IFC model

semhustej, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening IFC model (File -> Open) and using this import setup:
[image]

I get error messages. The file seems to open, at first sight it looks OK, but I did not dig deep whether something is missing.

Error message:

```
2:15:55  PropertyPythonObject::toString(): failed for <class 'nativeifc.ifc_viewproviders.ifc_vp_group'>
12:15:55  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\bin\Lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\bin\Lib\json\encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\bin\Lib\json\encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38830-conda-Windows-x86_64-py311\bin\Lib\json\encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
12:16:17  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:17:38  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:17:38  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:17:38  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:17:38  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:17:38  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:17:40  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:17:40  Imported residential_building_unnamed_4x0.ifc ( 7.88 Mb ) in 31:53
```

The file opens without errors in BonsaiBIM and multiple IFC viewers.

Dataset: 
[residential_building_unnamed_4x0.zip](https://github.com/user-attachments/files/17233142/residential_building_unnamed_4x0.zip)

WARNING: opening of the file takes 30 minutes on my PC. It opens in ca. 10 seconds in BonsaiBIM.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-10-03:
It opens OK with me in "Only root object" mode, in 7 seconds:

[image]

In "All individual objects" mode it just took too long and I killed it. I guess there are just too many objects created... I would recommend not using that mode, and just go expanding as needed...

Roy-043, 2024-10-03:
The Invalid iterator problem also occurs if you use "Only root object" and then double-click "Surface:1372221" and "_IfcBuilding". Everything except the site then becomes invisible which seems strange?

After adding `print(elements)` at the start of the `get_geom_iterator` function I get the following when expanding the building:
```
14:41:52  [#82643=IfcGrid('3nJTqVxzjattuxsS19eSHR',#18,'Default Grid',$,$,#82642,#82641,(#82634),(#82638),$,$)]
14:41:52  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
14:41:52  [#82655=IfcGrid('20EaNB78gKYPKX6GxICNC4',#18,'Default Grid',$,$,#82654,#82653,(#82646),(#82650),$,$)]
14:41:52  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
14:41:52  [#82667=IfcGrid('2AhcbJF2b3mXSOHY7GoCpM',#18,'Default Grid',$,$,#82666,#82665,(#82658),(#82662),$,$)]
14:41:52  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
14:41:52  [#82679=IfcGrid('3rl2m9$sxg3LiA4u3vVEeO',#18,'Default Grid',$,$,#82678,#82677,(#82670),(#82674),$,$)]
14:41:52  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
14:41:52  [#82691=IfcGrid('2sCTG7wBWM1VWjmJTLgGFT',#18,'Default Grid',$,$,#82690,#82689,(#82682),(#82686),$,$)]
14:41:52  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
14:41:52  [#82703=IfcGrid('0AixpYOE8OIO7HaNkHpzj2',#18,'Default Grid',$,$,#82702,#82701,(#82694),(#82698),$,$)]
14:41:52  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
14:41:52  [#82715=IfcGrid('3vxFdIL5rosLT2Qhn4dgIy',#18,'Default Grid',$,$,#82714,#82713,(#82706),(#82710),$,$)]
14:41:52  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
14:41:52  [#154=IfcSite('1cqAFKIjLCtv$f2mol$1KK',#18,'Surface:1372221',$,$,#153,#151,$,.ELEMENT.,(48,58,0,413818),(14,28,36,12954),0.,$,$)]
```

```
>>> import ifcopenshell
>>> ifcopenshell.version
'0.7.0'
```

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

yorikvanhavre, 2024-10-04:
We often have "invalid iterator" issues within IfcOpenShell. It's hard to know what causes that, and most of the time it does not affect the model. In this case though, it seems to be the Site object, but I'm not sure there s an actual problem there either, the model looks just fine and identical as what appears in other IFC viewers...

semhustej, 2024-10-05:
> The Invalid iterator problem also occurs if you use "Only root object" and then double-click "Surface:1372221" and "_IfcBuilding". Everything except the site then becomes invisible which seems strange?

This happen to me too. It is in fact a little strange.

When you double click one of the levels, the elements on that level appear.

semhustej, 2024-10-05:
> It opens OK with me in "Only root object" mode, in 7 seconds:
> 

> In "All individual objects" mode it just took too long and I killed it. I guess there are just too many objects created... I would recommend not using that mode, and just go expanding as needed...

Yes, loading Only root object works. 

I thought this model is not that big to overwhelm FreeCAD when loading/displaying all geometry. It is not too detailed. It is Revit export, so I guess it is far from ideal IFC file, that might be the culprit.

semhustej, 2024-10-05:
I tried finding the culprit of the error messages. So I used Bonsai to extract each class of IfcElement in the file to a separate file:
[decomposed.zip](https://github.com/user-attachments/files/17266002/decomposed.zip)

Then I proceeded to open these in FreeCAD using these settings:
[image]

All of the files imported without errors (only DEBUG information). When adding all load times, the loaded under 6 minutes (which is way less than the original file loading in over 30 minutes for me):

```
12:22:22  Imported extract-BuildingElementProxy.ifc ( 0.44 Mb ) in 00:09
12:26:00  Imported extract-door.ifc ( 2.53 Mb ) in 03:23
12:26:53  Imported extract-furinture.ifc ( 0.55 Mb ) in 00:00
12:27:03  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:27:03  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:27:03  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:27:03  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:27:03  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:27:03  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:27:03  DEBUG: ifc_tools.get_geom_iterator: Invalid iterator
12:27:03  Imported extract-grid.ifc ( 0.02 Mb ) in 00:00
12:27:25  Imported extract-railing.ifc ( 0.55 Mb ) in 00:06
12:27:36  Imported extract-slab.ifc ( 0.18 Mb ) in 00:01
12:27:53  Imported extract-stair.ifc ( 0.05 Mb ) in 00:00
12:28:03  Imported extract-stair-flight.ifc ( 0.15 Mb ) in 00:01
12:28:48  Imported extract-wall.ifc ( 2.56 Mb ) in 00:38
12:30:13  Imported extract-window.ifc ( 0.77 Mb ) in 01:15
```

It does not seem to me, that the problem with the original file is with too much or too complicated geometry or data. 

As stated above, the original file is Revit export, so far from optimal ifc file. If it is not worth investigating, the issue can be closed.

yorikvanhavre, 2024-10-08:
> The Invalid iterator problem also occurs if you use "Only root object" and then double-click "Surface:1372221" and "_IfcBuilding". Everything except the site then becomes invisible which seems strange?

You need to recompute all the new objects created under _IfcBuilding. It's a known bug, and I haven't found a good solution to it yet :/ There is a workaround in https://github.com/FreeCAD/FreeCAD/pull/16629 though, but it needs further testing

> All of the files imported without errors (only DEBUG information). When adding all load times, the loaded under 6 minutes (which is way less than the original file loading in over 30 minutes for me)

That is because by importing these 10 files, you create 10 objects only. The problem in FreeCAD is not the complexity of the model, it's the number of individual objects.

In any case, this problem (too many objects in the FreeCAD document) is not fixable at the moment, so I don't think there is much we can do here... I'm going to close the issue, okay? Don't hesitate to reopen if you feel the need for it

closed 2024-10-08 by yorikvanhavre