FreeCAD/FreeCAD #16971: can't scale smaller than 0.01

foobarbecue, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new ["Part Scale" feature](https://wiki.freecad.org/Part_Scale) is great, but it doesn't let you scale to smaller than 0.01x because the input box only allows two digits after the decmial place. I wanted to use it to convert a part from millimeters to centimeters so needed 0.001x.

I would be interested in fixing this myself since it seems like a decent bite-size first-time contribution.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-02:
Right, that field doesn't support the expression editor so you would have to increase the number of decimals globally.

Roy-043, 2024-10-02:
You can edit the scaling in the Property editor afterwards. You can use an expression then.

For Draft_Scale task panel we use this:
https://github.com/FreeCAD/FreeCAD/blob/813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6/src/Mod/Draft/drafttaskpanels/task_scale.py#L50

mwganson, 2024-10-02:
The reason there is no expression support in the dialog is you need an existing object with existing properties in order to bind the dialog widget to it, and in this case, as is the case for most Part WB dialogs, the object is only created upon closing the dialog.  This also prevents getting a preview in the dialog of the completed object.

Contrast this to Part Design where you get the preview and the expression binding in the dialog.  If you switch to the model tab while the dialog is open you will find the object is already created while the dialog is open.  If you switch back to the dialog and cancel, then this object gets removed.

It is probably relatively simple to change the number of digits allowed in the dialog using the c++ equivalent of something similar to Roy's suggestion (maybe even just by editing the ui file by hardcoding the number of digits to use), but adding expression binding would be quite a bit more complicated.

Personally, I have my global setting to use 6 digits of precision.

foobarbecue, 2024-10-02:
Aha, I didn't know there was a setting for "Number of decimals" . That fixes the problem nicely.

A way to make this user-friendly for new users would be to trigger a popup the first time a user tries to enter a number that exceeds default precision, telling them that there is a way to set the precision. That might be difficult to implement, I guess.

luzpaz, 2024-10-17:
> It should probably not be enforced on entering values, but rather just on how they are displayed, but I don't know how difficult or easy that would be or if even Qt would support it.

Should we generate a new ticket from this, in that case ?

mwganson, 2024-10-18:
I'm not sure about the ticket question.  The more I think about it, it's probably better to leave it as is.  We don't want to confuse the user into thinking more decimals of precision are being used than actually are being used.  Already there are 2 good options: 1) change the system decimals setting in preferences; or 2) use the expression engine to add more digits of precision.  In the latter case, only 2 digits (for example) would be shown, but internally the full value would be used.

foobarbecue, 2024-11-16:
Thanks for the discussion everyone -- sounds like we can close this as not planned. At least this issue will serve as a bit of documentation for anyone who wonders about this.

closed 2024-11-16 by foobarbecue