FreeCAD/FreeCAD #16972: Assembly: can no longer create BOM

furgo16, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD
2. On the start page, click on the assembly example shipped with FreeCAD (backhoe) to open it
3. Double-click on the assembly in the document tree to activate it
4. Click on the BOM command
5. Failure with the following error on the report window:

```python-traceback
Running the Python command 'Assembly_CreateBom' failed:
Traceback (most recent call last):
  File "/snap/freecad/1153/usr/Mod/Assembly/CommandCreateBom.py", line 94, in Activated
    self.panel = TaskAssemblyCreateBom()
  File "/snap/freecad/1153/usr/Mod/Assembly/CommandCreateBom.py", line 146, in __init__
    self.createBomObject()
  File "/snap/freecad/1153/usr/Mod/Assembly/CommandCreateBom.py", line 335, in createBomObject
    Gui.doCommand(commands)
  File "<string>", line 1, in <module>

name 'assembly' is not defined
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38844 (Git) Snap 1153
Build type: Release
Branch: main
Hash: 6e2cd4e73303f817e4bf5e908abe39030b7c5fb3
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-02:
Can't reproduce. This is what I see when clicking on BOM button

[image]

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```

furgo16, 2024-10-02:
@luzpaz thanks for testing. I could reproduce it on the snap package, on the PPA package and on a local build. I've added a screencast and a workaround to the issue description, along with a PR.

furgo16, 2024-10-02:
@luzpaz looking at your spreadsheet, it seems you skipped step 3, which is the key step to trigger the bug.

luzpaz, 2024-10-02:
I think I got confused because the ticket was being edited while I was checking.

closed 2024-10-07 by yorikvanhavre