FreeCAD/FreeCAD #16974: Assembly: Assembly example shows original components and assembly links

furgo16, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Start FreeCAD
1. On the start page, click on the assembly example shipped with FreeCAD (backhoe) to open it
1. Press <kbd>V, F</kbd> to fit all objects to the view
1. Expected: only the assembly is shown
1. Actual: the assembly and its individual parts are shown

[image]

I would expect that the `CAD` group in the document tree is set to hidden, not visible. However, if this is intended, please feel free to close this issue.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38844 (Git) Snap 1153
Build type: Release
Branch: main
Hash: 6e2cd4e73303f817e4bf5e908abe39030b7c5fb3
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-03:
You can create a PR with an updated example file.

furgo16, 2024-10-03:
Happy to, I first wanted to know if it's the intended behavior.

maxwxyz, 2024-10-03:
be sure that save thumbnail is activated and watermark logo is disabled when saving the updated file (also please in Classic theme) so that the preview image is in line with the other examples:
[image]

luzpaz, 2025-12-25:
@furgo16 have at it

furgo16, 2026-09-19:
This is still relevant. The fix is trivial: mark the `CAD` group in the file as hidden by selecting it and pressing space, and then resave the file.

I was saving this as a good first issue for a first time contributor, but given that's been open for nearly two years, I might just submit the fix myself. Or alternatively, @maxwxyz I saw you were updating the example files for 26.3 in a PR - are you done with the changes, or could you add this fix too?

maxwxyz, 2026-09-19:
@furgo16 this change was already included in my update

furgo16, 2026-09-19:
Excellent, thank you.

closed 2026-09-27 by Lgt2x