FreeCAD/FreeCAD #16975: TechDraw: Label placement is typographically wrong

pierreporte, 2024-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When snapping dimensions together to form a continuous line, the line isn’t continuous if one of the label has a different height, as described in issue #16954. However, when both labels are a single line without anything special like tolerances, they can still have a different height if one is only a number and the other has characters with descenders, hence making the two dimensions misaligned as illustrated below.

[image]

It means that the position of the label on the dimension line is calculated with the bounding box of the text. In typography, positioning text is more complex than using bounding boxes. It generally uses the base line, which has a constant distance to a reference line (here the dimension line), but it can also use the point size, in which the base line is positioned, which is known. See the illustration below. (By the way, length 13 corresponds to the blinking caret in text editors.)

[image]

If the dimension labels were position such as the base line is always X mm from the dimension line, or the descender line is Y mm from it, then the labels will be aligned when #16954 is fixed. In the current state, even if this issue was fixed, alignment problems would still be present.

Also, text decorations like frames (for theoretical dimensions) and ISO 129-1 symbols (requested in #13617) should not disturb text alignment and the base line of their text should be at the same position as surrounding text.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-03:
@benj5378 maybe you want to give this a try?
@PaddleStroke FYI

WandererFan, 2024-10-03:
Why is this different from #16954? 

@maxwxyz is there a way to merge issues?  There is info in each issue that is not in the other.

maxwxyz, 2024-10-03:
@WandererFan unfortunately not. It has to be add as comment manually

pierreporte, 2024-10-03:
I hesitated to add this in the other issue but eventually thought it was different enough. Even if the alignment of dimensions was calculated with lines instead of text, the text itself can be misaligned. Granted, both issues are very close.

WandererFan, 2024-12-28:
Note that the fix for issue #11452  (PR #18771) seems to correct the wrong alignment with comma decimal separators and descenders.

[image]

closed 2025-01-13 by chennes

pierreporte, 2025-01-13:
Quoting @WandererFan in https://github.com/FreeCAD/FreeCAD/pull/18771#issuecomment-2571671439

> The difficulty for us is that Qt uses the top left of the text item's bounding rectangle (which is larger in every direction around the text) as the position point, but does not provide a method of determining where the glyphs are within that rectangle.
>
> We could use something like FreeType to find all sorts of numbers related to the font, but event that doesn't help us if we don't know where Qt puts the text within the bounding rect.
>
> Maybe we need to look inside QGraphicsText item code to see how it determines the brect.

and picture he posted:

[image]

It’s not sure that the problem has been fully solved because we can’t access the real base line of text.

@maxwxyz Should the issue be re-opened or should it remained closed and new one created if we found new issue down the line?

maxwxyz, 2025-01-15:
IMO a new issue

benj5378, 2025-05-07:
> Note that the fix for issue https://github.com/FreeCAD/FreeCAD/issues/11452 (PR https://github.com/FreeCAD/FreeCAD/pull/18771) seems to correct the wrong alignment with comma decimal separators and descenders.

This is still an issue.

> Maybe we need to look inside QGraphicsText item code to see how it determines the brect.

I did that, taking hopelessly long, it transcends through 5 layers to `QPlainTextDocumentLayout::documentSize()` or something like that. I found a possible solution even [StackOverflow](https://stackoverflow.com/a/79610699/7537127) worthy.

All these issues can be solved by making text placement relative to font baseline (instead of bounding box) or using my above linked solution.