FreeCAD/FreeCAD #16978: TD + BIM Section plane. Render mode affects rotation in the drawing.

kaiwas, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the display mode of the cutting plane on the drawing changes, the element rotates.
The entire sequence of actions from creation to modification is shown in the video.

https://github.com/user-attachments/assets/7b70a835-608c-4097-9931-16ae2d3a8be8

The file for tests is attached (Remove the zip extension)
[TD_render_Mode.FCStd.zip](https://github.com/user-attachments/files/17240389/TD_render_Mode.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
  * lattice2 1.0.0
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-10-03:
That's indeed a known problem: The OpenCasCade HLR (hidden lines removal) algorithm, that is used for wireframe mode, only takes a projection direction, and no Up Vector. Without the up vector, it's not possible to determine the rotation of the projection around its projection axis.

The Solid projection mode uses our own code, and there we can use an up vector and therefore control the rotation.

I don't see an easy solution to this at the moment, unfortunately... We could build some code to "analyze" the HLR results and rotate if needed, but that would be a lot of work for something very error-prone.

kaiwas, 2024-10-03:
> I don't see an easy solution to this at the moment, unfortunately... We could build some code to "analyze" the HLR results and rotate if needed, but that would be a lot of work for something very error-prone.

I think this is not such a critical problem that it would be worth solving it with great efforts in the near future. Thank you!

yorikvanhavre, 2024-10-03:
Let's leave this open so maybe someone gives an idea of a good solution here...

FEA-eng, 2025-10-01:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```