FreeCAD/FreeCAD #16979: FEM: Incorrect constraint symbols and mesh for rectangular ruled surface on the XZ plane

FEA-eng, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Originally reported here: https://forum.freecad.org/viewtopic.php?t=91105

1. Sketch a rectangle on the XZ plane (it works normally on XY).
2. Use Part Ruled Surface on the 2 long edges. Hide the Sketch.
3. Apply fixed constraint, force and mesh. All are incorrectly displayed:

[image]

According to the OP, it happened in 0.21 too. It doesn't seem to happen with Part Make face instead of Ruled surface.

[FEM ruled surface.FCStd.zip](https://github.com/user-attachments/files/17243587/FEM.ruled.surface.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-03:
@marioalexis84 can I assign this ticket to you ?

marioalexis84, 2024-10-03:
The problem is not FEM related, it occurs directly on the ruled surface when the sketch is not on the xy plane.
Try this example by creating the ruled surface between edge1 and edge3 of the sketch.

[fail_ruled_surface.FCStd.gz](https://github.com/user-attachments/files/17250071/fail_ruled_surface.FCStd.gz)

FEA-eng, 2024-10-04:
I guess that's related then: #15675

luzpaz, 2024-10-05:
@FEA-eng should we close as dupe ?

FEA-eng, 2024-10-05:
> @FEA-eng should we close as dupe ?

That older issue mentions only offset. Maybe instead of those two issues there should be a general one like "Part Ruled surface is incorrect on the XY plane" listing the two reported problems.

FEA-eng, 2024-10-14:
Closing in favor of #17235

closed 2024-10-14 by FEA-eng