FreeCAD/FreeCAD #16980: Problem with hidden views when scaling TechDraw

picli3, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a flat piece with a circle inside that has rounded edges.

[image]

When I project the faces onto a technical sheet, these roundings look like complete circles.

[image]

I have seen that this is corrected if I increase the scale and it gets worse if I decrease it.

[image]

I hope it's easy to solve. Thanks for all the work you've done in FreeCad

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Peru (es_PE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.66
  * kicadStepUpMod 11.2.9
  * ExplodedAssembly
```

### Subproject(s) affected?

None

### Anything else?

[Sample piece](https://drive.google.com/file/d/19p_rleQw1MOGd0f_j31UDd9Qkb-4zC_E/view?usp=sharing)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-03:
can confirm.
@WandererFan FYI

benj5378, 2024-10-03:
Related til #16898?

commit 42a61ca8 (2024-10-14): [TD]Fix bad circles (fix #16980) (#17182)

* [TD]fix spurious messages in GeometryMatcher

* [TD]fix closed circle test (fix #16980)
commit 010c3c9f (2024-10-14): [TD]Fix bad circles (fix #16980) (#17182)

* [TD]fix spurious messages in GeometryMatcher

* [TD]fix closed circle test (fix #16980)
closed 2024-10-14 by yorikvanhavre