FreeCAD/FreeCAD #16981: "ReferenceError: Cannot access attribute 'ActiveView' of deleted object" when rotating camera

KrisSlyka, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening up carriage.FCStd from the attached zip rotating the camera causes the following error to appear over and over in the report view.
Deleting the object named "Rod Tilted" in the subassembly included from suspension.FCStd makes the error go away. It does not appear when opening suspension.FCStd directly, only when it's included as a subassembly.

```
14:00:34  Traceback (most recent call last):
14:00:34    File "/usr/lib/freecad/Mod/Assembly/JointObject.py", line 1202, in camera_callback_rotation
14:00:34      self.set_lock_rotation()
14:00:34    File "/usr/lib/freecad/Mod/Assembly/JointObject.py", line 1205, in set_lock_rotation
14:00:34      activeView = get_active_view(self.gui_doc)
14:00:34                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
14:00:34    File "/usr/lib/freecad/Mod/Assembly/JointObject.py", line 166, in get_active_view
14:00:34      activeView = gui_doc.ActiveView
14:00:34                   ^^^^^^^^^^^^^^^^^^
14:00:34  ReferenceError: Cannot access attribute 'ActiveView' of deleted object
```

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * freecad.gears 1.3.0
  * fasteners 0.5.29
  * OpticsWorkbench 1.0.24
  * fcgear 1.0.0
```

### Subproject(s) affected?

Assembly

### Anything else?

[carriage.zip](https://github.com/user-attachments/files/17243899/carriage.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-03:
@PaddleStroke @Rexbas any idea?

PaddleStroke, 2024-10-04:
For some reason this document is being partial loaded again. When you open it you get this warning : 

> 0 <App> Document.cpp(2130): Reload partial document suspension

So what happens is that when the fixed joint in 'suspension' doc is loaded, the activeView of suspension document is not none and we save gui_doc to later access it in the camera callback. However since the document is being reloaded for some reason, then the gui_doc that we saved is now bad. So when the callback tries to use it it's giving this error.

Now the question is, why this document is being reloaded. It must have something wrong

furgo16, 2024-10-04:
I can also trigger this error when opening an assembly that contains external components, and manually reloading any of those external components via right-clicking them. I happened to be testing https://github.com/FreeCAD/FreeCAD/issues/17006, it then happened on a larger assembly I was working on and it really confused me. Thanks a lot for reporting this issue, I'm glad I found it.

closed 2024-10-14 by chennes