FreeCAD/FreeCAD #16983: Placement (Rotation) using expressions has no "connection" with the object's property

kaiwas, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If we set up rotation using the Placement tool with Euler angles, we do not see in the object's property that such a connection exists. At the same time, we can change the values ​​in the property (but they do not change to the values ​​we enter), while the values ​​in Placement remain unchanged.

Video demonstration. 

https://github.com/user-attachments/assets/4ccb6233-4a74-41a3-84d5-232d4df5a816

The test file is attached (remove the zip extension)
[Placement_EXP.FCStd.zip](https://github.com/user-attachments/files/17245509/Placement_EXP.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * lattice2 1.0.0
  * PitchedRoof
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-03:
Just FYI: the Euler angle expressions are (of course) listed  in the "Expression Engine" property (which is hidden by default).

kaiwas, 2024-10-03:
> Just FYI: the Euler angle expressions are (of course) listed in the "Expression Engine" property (which is hidden by default).

Yes. Inconveniently, you can see that Euler angles are used. (See, but do not change)
But the quaternion input fields are still active and lead to an incomprehensible result. I think this is not quite correct.

luzpaz, 2024-10-30:
@PaddleStroke is this something that is similar to the recent fix you made in Assembly (https://github.com/FreeCAD/FreeCAD/pull/17406) ?

PaddleStroke, 2024-11-04:
Not really. I'm not sure how to do this since the object properties seems to be the quaternion.
@wwmayer this is a core issue that may interest you.

kaiwas, 2025-10-01:
> For some reason, when I open this file, "Euler" is renamed "0" and has empty Data properties:

I noticed that too.
The name is renamed to "Zero."
But the data seems to be there.

[image]

kaiwas, 2026-10-19:
still be relevant