FreeCAD/FreeCAD #16987: [Spreadsheet] Cannot finish expression before cell entering exit

benj5378, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open spreadsheet in empty document
2. Try to enter `=263346`
3. Observe that text entering is exited upon entering `=26`

Only happens for some specific numbers, often in ranges. Expected behaviour is that I can finish my expression before the cell is deselected/exited by itself

This might be a Wayland specific bug

### Full version info

```shell
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Debug
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-03:
Can't confirm.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

xtemp09, 2024-10-03:
Everything works as expected.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38842 (Git)
Build type: Debug
Branch: master
Hash: 69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

----

@benj5378, do you observe stealing of selection with other applications?

hyarion, 2024-10-03:
<details><summary>Cannot confirm on macOS either</summary>
<p>
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38888 (Git)
Build type: Release
Branch: main
Hash: d423795e2dc32dc16f1e8c1835104deb01c9cab9
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```
</p>
</details>

amreo, 2024-10-03:
I cannot reproduce on Manjaro.

```
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * ExplodedAssembly
  * sheetmetal 0.4.24
```

benj5378, 2024-10-03:
As I wrote, I think this is a Wayland issue. So please specify if you're on Wayland or not.

> @benj5378, do you observe stealing of selection with other applications?

@xtemp09, no, it's not an issue in other applications. Not an issue in LibreCalc. Not an issue in FreeCAD snap edge. Only an issue on my personal Wayland build.

xtemp09, 2024-10-04:
@benj5378, does it mean that when you run `QT_QPA_PLATFORM=xcb ./FreeCAD` it works fine?

benj5378, 2024-10-04:
> @benj5378, does it mean that when you run `QT_QPA_PLATFORM=xcb ./FreeCAD` it works fine?

That is correct, running with your command, the issue isn't present. @xtemp09

xtemp09, 2024-10-04:
Perhaps, these issues are relevant:

https://gitlab.gnome.org/GNOME/mutter/-/issues/2868

https://gitlab.gnome.org/GNOME/mutter/-/issues/2690

tomunger, 2024-12-28:
I can confirm this problem on Mac OS in the most recent and previous weekly builds.

It seems that typing any equation will cause the cell to stop taking input after 2 or 3 keystrokes.  Currently I must type in the equation then edit "=" to the front of the text.

I've included a file with a spreadsheet

- In cell B5, typing "=this+that" results in, well, not able to complete the text entry.
- in cell B5, typing "=1234" results in same.

`OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * DynamicData 2.70.0
  * Manipulator 1.5.0
  * fasteners 0.5.21
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Assembly3 0.12.0
`

[freecad_spreadsheet_bug.FCStd.zip](https://github.com/user-attachments/files/18267699/freecad_spreadsheet_bug.FCStd.zip)

hyarion, 2024-12-28:
@tomunger I think this is a new regression, I can see this issue on latest main branch too, but on 38871 and 1.0.0 it wasn't an issue on macOS.
@xtemp09 could this be a regression caused by the zoom feature?

hyarion, 2024-12-28:
@tomunger workaround is to use the Content field above the spread sheet to enter expressions

tomunger, 2024-12-28:
This was not a problem in a mid-November weekly build.  It is something new.  I've not tested 1.0

xtemp09, 2024-12-29:
@hyarion, I don't know, but the issue was reported long before my PR.

When I was making the zoom, I expected trouble with trackpad/trackball/spacemouse/gesture systems.

tarman3, 2024-12-29:
Reproduced in FreeCAD 1.1.0dev.39741 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)
Not reproduced in FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)

Perhaps it has something to do with autocomplete

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.39741 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

xtemp09, 2024-12-29:
I think that this issue and the issue #13863 result from fiddling about focus events. Overlay manager/floating widgets/event filters are likely to cause them both.

MattN92705, 2025-02-15:
I think I am seeing this same bug (or something very similar) in the last couple of dev versions on MacOS

I create an alias for a cell, e.g., "myAlias", then in any other cell, if I attempt to enter any expression using that alias, e.g., "=myAlias+1", the text entering is exited as soon as I type "=m".  leaving only " '=m " in the cell (with that annoying single quote).

This happens with a fresh document with nothing but a new spreadsheet in it.

Fortunately there exists a workaround, as typing into the "Content:" field works as expected.

OS: macOS 15.3
Architecture: arm64
Version: 1.1.0dev.40253 (Git) Conda
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Logical/physical DPI: 72/108
Installed mods: 
  * OpenTheme 2024.9.1

xtemp09, 2025-02-17:
@benj5378, could you check if this commit fixes the issue?

https://github.com/xtemp09/FreeCAD/tree/iss-16987

xtemp09, 2025-02-23:
I have just checked the latest weekly bundle and failed to reproduce the problem.

Is it the right time to close the issue?

MattN92705, 2025-02-23:
I just checked dev version 40317 - the problem I mentioned above appears to be resolved.  Thanks guys!

benj5378, 2025-02-23:
I cannot reproduce it at least

closed 2025-02-23 by benj5378