FreeCAD/FreeCAD #16988: Unable to specify circle radius in sketch

Beep6581, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://filebin.net/1r0fvn7t6ahztl1w

1. Open attached project.
2. Double-click Lid > Pocket004 > screw holes
3. Delete 1-circle to 4-circle
4. Create circle by center, click on the centers of the holes, e.g. on the top-right hole's center: voltage_regulator_box_2.Body001.Pocket004.Sketch007.Vertex1
5. Type "3,5" for a 3,5mm diameter circle. It creates a circle exactly the same size as the construction geometry, not 3,5mm.

If you restart FreeCAD and repeat these steps, except in step 5 you type "6" instead of "3,5", then it works. It seems to not like the fraction.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenDark 2023.12.17
  * fasteners 0.5.25
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.9
  * Curves 0.6.42
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-03:
It sounds like freecad uses a number format that you don't expect.

Can you check if writing `3.5` works?

Also which number format is used in https://wiki.freecad.org/Preferences_Editor#General_settings ?

amreo, 2024-10-03:
[image]
I cannot reproduce, even in my Locale the comma is used for decimal separator

maxwxyz, 2024-10-04:
cannot reproduce on 
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

Do you have an old config file on the system from 0.21.2? Maybe rename your user.cfg and try again.

Beep6581, 2024-10-05:
```
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=sv_SE.UTF-8
LC_TIME=sv_SE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=sv_SE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=sv_SE.UTF-8
LC_NAME=sv_SE.UTF-8
LC_ADDRESS=sv_SE.UTF-8
LC_TELEPHONE=sv_SE.UTF-8
LC_MEASUREMENT=sv_SE.UTF-8
LC_IDENTIFICATION=sv_SE.UTF-8
LC_ALL=
```

> Can you check if writing 3.5 works?

From memory, it did not.

> Also which number format is used in https://wiki.freecad.org/Preferences_Editor#General_settings ?

"Operating system".

> Do you have an old config file on the system

No.

```
$ locate user.cfg
/home/bob/.config/FreeCAD/user.cfg
/home/bob/.local/share/FreeCAD/SavedPreferencePacks/user.cfg.backup
```

[image]

I updated to the latest AppImage and cannot reproduce.

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenDark 2023.12.17
  * fasteners 0.5.25
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.9
  * Curves 0.6.42
```

You can close, and I will report if I encounter it again.

closed 2024-10-06 by Roy-043