FreeCAD/FreeCAD #16991: Measurement: It’s not possible to project in arbitrary LCS or on geometry

pierreporte, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new measurement tool can now show deltas using the origin coordinate system. It is not possible to select an arbitrary LCS to show the deltas in it. It is also not possible to project a measurement on a plane, a surface, or a line.

Expected projection results:

| Original measure | Target           | New measure           |
| ---------------- | ---------------- | -----------           |
| Length           | Plane or surface | Length, possibly zero |
| Length           | Line             | Length, possibly zero |
| Area             | Plane or surface | Area, possibly zero   |
| Area             | Line             | Length, possibly zero |

### Full version info

```shell
OS: Ubuntu 24.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git) AppImage
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```

### Subproject(s) affected?

Core

### Anything else?

Follow-up of #13741

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-06:
How would you distinguish geometry to be measured from geometry to be projected onto?
What should be calculated if the user selects a line and a face? The shortest distance between the elements, or the the length of the edge projected onto the face?

pierreporte, 2024-10-06:
The process is actually in two steps. First, the user makes a measurement, whatever it is. Then they click on something in the task view which allows them ti select the projection target the measurements will be projected in.