FreeCAD/FreeCAD #17005: Display of Variable based constraints

OnlineDynamic, 2024-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dialog boxes related to constraints don't make it very clear if the constraint is set to a hard coded value or referencing an expression ( or variable set/spreadsheet named value)

It would be great if the expression/refernced variable name was displayed and seperately show its current calculated value in the dialog rather than having to hover over the tiny expression icon.

I know there is already an open issue for fixing the referenced values being lost when copying and pasting objects - this change will help make it a lot clearer if the constraint isn't tied to an expression

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.48
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-10-04:
Duplicate of #12036

maxwxyz, 2024-10-04:
Good idea, do you have an idea or concept to show with a screenshot?
Loosely related: https://github.com/FreeCAD/FreeCAD/issues/12853

luzpaz, 2024-10-24:
@OnlineDynamic :point_up_2:

OnlineDynamic, 2024-10-24:
could be as simple as displaying the expression/refernced variable name in the existing dialog e.g.
VariableSet('variableName') (set current value in brackets)

[image]

so instead of just showing 12.40mm in the diamter box show VariableSet('myvariableName') (12.40mm)