FreeCAD/FreeCAD #17006: Assembly: document needs to be manually reloaded to show grouping of external components

furgo16, 2024-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Unzip the [test file](https://github.com/user-attachments/files/17257229/A0.zip)
2. Open the A0.FCStd file with FreeCAD
3. Click on the arrow left to the A1 file on the document tree to expand it
4. Notice that there are no groups in the file
5. Right-click on the A1 file on the document tree and select "Reload document"
6. Now the group from the original file is shown

After opening the file there are no groups in A1:
[image]

Groups in A1 are only shown after manually reloading the document:
[image]

 

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38844 (Git) Snap 1153
Build type: Release
Branch: main
Hash: 6e2cd4e73303f817e4bf5e908abe39030b7c5fb3
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-05:
I cannot reproduce. This is how it looks after opening without recompute:
[image]

Recomputing both documents does not change anything.
How would this issue be related to assembly WB?

furgo16, 2024-10-05:
Thanks for testing.

> How would this issue be related to assembly WB?

I'm not an expert, so happy to be corrected here. My assumption was that the Assembly workbench loaded the external components from the main assembly.

Roy-043, 2024-10-06:
Confirmed. This is called 'partial loading'. It is related to using external Links.
Some background info can be found here:
https://github.com/realthunder/FreeCAD_assembly3/wiki/Link

furgo16, 2024-10-06:
Thanks. Following the links on that URL, this doc describes partia document loading: https://github.com/realthunder/FreeCAD_assembly3/wiki/Core-Changes#partial-document-loading, although I cannot tell how much of it has been included in main, and how much is RT-branch-specific.

For instance, I'm not sure if the quote below applies:

> The tree view shows a grey icon for partial document. The user can double click the grey icon to reload the document fully

If I undrestand it correctly though, it's about recomputation and dependecy loading. I still cannot tell if groups should be loaded, though.

Finally, is that related to this preference setting as well?

[image]

I have it unchecked on my system, so if it is, partial loading should take place.