FreeCAD/FreeCAD #17007: Bulk Edit of Constraints

OnlineDynamic, 2024-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great if you were able to bulk edit multiple constraints at the same time.

I'm thinking if the user selects multiple constraints of the same type they ought to be able to right click and have a bulk edit option.

This would bring the normal dialog for editing the constraint but when applied it would apply across all originally selected constraints rather than just one

My current use case - creating lots of circles which all need to be constrained to the same diameter (in my case value held in a variable set)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.48
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-04:
workaround would be to make them equal or use expressions

FlachyJoe, 2024-10-04:
Using many datum constraints with the same value is a bad practice. Please use [equality](https://wiki.freecad.org/Sketcher_ConstrainEqual).

OnlineDynamic, 2024-10-04:
> Using many datum constraints with the same value is a bad practice. Please use [equality](https://wiki.freecad.org/Sketcher_ConstrainEqual).

I would agrue creating 100 hundreds of equal constraints gets messy as its hard to work out the master reference object - much better all constraints are tied to a named variable value

OnlineDynamic, 2024-10-04:
> It's not more complicated to select all the circles in the geometry list than all the datums in the constraint list. Just press `E` when they are all selected _et voilà_!

That might be a quick way to add the equality constraint but still doesn't help usability going forwards - as all the objects are inter related - try it on a thousand circles - now all cross referencing each other - thats a mess - much better if each one just had a tie to a single variable.  Currently if you delete one object in that string of equality constraints hundreds of downstream items loose the reference to the original constraint

FlachyJoe, 2024-10-04:
So the issue is the geometries has to be equal to the first selected and not create a chain.

kaiwas, 2024-10-04:
> you certainly miss something, a sketch with so many circle is unusual, can't you use [a Point Array](https://wiki.freecad.org/index.php?title=Draft_PointArray)?

As it often happens - you project your approaches onto all possible others. Thinking that your approach is the only correct one.
Maybe this is one of the necessary features - it is possible to remove a part from a large number.
Tasks can be completely different and I also miss multiple operations.
There should be an opportunity to both maintain strong connections and have the ability to destroy them when necessary.

FlachyJoe, 2024-10-04:
@kaiwas you can put screw with a hammer, it works. But it's not optimal and it's not what the creator of the hammer like you do with it. I'm one humble hammer creator.

kaiwas, 2024-10-04:
> @kaiwas you can put screw with a hammer, it works. But it's not optimal and it's not what the creator of the hammer likes you do with it. I'm one humble hammer creator.

And I'm not calling for solving something here and now. But it would be nice to admit that such a possibility could be useful (in some cases).

The first thing that came to my mind was about equalities. It's the simplest. But then I remembered that if I want to change something, it will destroy everything. For now, I have to run through all the constraints and press Ctr+V to insert an expression.

I'm not trying to belittle your merits in creating the hammer, but just stop thinking that the roundabout paths are always correct.

FlachyJoe, 2024-10-04:
Many functionalities can be useful but we don't have the capability to answer all the requests so please try the tools you currently have in the box and if you don't find any to do the job then ask more. As myself you certainly don't know all the possibilities and the features.

I don't promote workarounds but just choose the tool we already done for the task if it exists.

kaiwas, 2024-10-04:
> I don't promote workarounds but just choose the tool we already done for the task if it exists.

Sorry if I misunderstood your words.
It's just that all sorts of workarounds are often perceived as "the specified problem is not important (just because there is some workaround)"