FreeCAD/FreeCAD #17017: Boolean Issue

CrippleCat, 2024-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you use a boolean with a large object and a smaller object, or when you use a boolean on a boolean object, then Freecad crashes. This is a reoccurring problem on my version (not a one off)

### Full version info

```shell
MAC OS non arm version 1.10dev-38871
```

### Subproject(s) affected?

None

### Anything else?

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [98066]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-04 19:57:53.9034 +0100
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Anonymous UUID:        473056DD-6AF5-8BEB-38E1-E31D82587080

Sleep/Wake UUID:       C17E1410-1D4A-42A4-8F7F-D9A0D4F86D93

Time Awake Since Boot: 1300000 seconds
Time Since Wake:       12016 seconds

System Integrity Protection: enabled

Notes:
PC register does not match crashing frame (0x0 vs 0x1384309A2)

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [98066]

VM Region Info: 0 is not in any region.  Bytes before following region: 4364034048
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1041de000-1041fe000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x1384309A2)

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   MeasureGui.so                 	       0x1384309a2 MeasureGui::QuickMeasure::addSelectionToMeasurement()::$_0::operator()(App::DocumentObject*) const + 34

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-04:
@johnocfii do you mind testing on macOS ? TIA

luzpaz, 2024-10-04:
@CrippleCat please always add your full [About](https://wiki.freecad.org/About) info. It helps us limit the back and forth troublshooting of 3rd party dependency related bugs. TIA

JohnOCFII, 2024-10-04:
> @JohnOCFII do you mind testing on macOS ? TIA

Tested with the fresh weekly, and I did not see a crash or error of any type.  I **am** using an Arm Mac -- not sure if that is critical to this issue.

John

```
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```
[image]

CrippleCat, 2024-10-04:
I was not using Arm, so that could be the problem, this is the version that I am using.
- [FreeCAD_weekly-builds-38871-conda-macOS-x86_64-py311.dmg](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-38871-conda-macOS-x86_64-py311.dmg)

luzpaz, 2024-10-05:
@CrippleCat again, please paste the full [About](https://wiki.freecad.org/About) info

luzpaz, 2024-10-05:
I **can't reproduce crash** but i do see this when I select the cylinder:  
```shell
04:26:18  <Exception> TopoShapeExpansion.cpp(1395): Null shape
04:26:18  Boolean: Null shape
```

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```

CrippleCat, 2024-10-05:
Here is a link to a video of what happens when I use a boolean on that file
https://drive.google.com/file/d/1vb2vuPBb9X7T1pUr3W1Yz3bC2Kdne-YT/view?usp=sharing

Here is the about info, sorry for not adding that earlier

OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/unset/Qt default
Installed mods: 
  * Help 1.0.3

luzpaz, 2024-10-05:
@CrippleCat can you start freecad from the terminal and then try to reproduce crash and see what the output is ? (copy/paste back here)

CrippleCat, 2024-10-05:
Here was the crash report given (on the link, cant paste all of it here), Im not sure if thats what you are after but nothing came up in the macos terminal. (I am quite new to this so sorry if this isnt what you need)
https://docs.google.com/document/d/1OMetejU3NtKfLg1d7FvuEbgg-5JHLAYFyfy-3kXRXt4/edit?usp=sharing

CrippleCat, 2024-10-05:
I have now just updated to the version down below (the latest weekly build), and everything seems to work fine, even with more complex models, so I presume maybe it was just something wrong with the download I had?

OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/unset/Qt default
Installed mods: 
  * Help 1.0.3

JohnOCFII, 2024-10-05:
> I have now just updated to the version down below (the latest weekly build), and everything seems to work fine, even with more complex models, so I presume maybe it was just something wrong with the download I had?

So weird.  I note in the video you seemed to drag the crash report into view.  Do you have more than one monitor attached?

CrippleCat, 2024-10-05:
I have 2, however freecad usually defaults to my other monitor, hence why that report came up on that. It all seems to be working fine now as nothing has crashed yet.

luzpaz, 2024-10-06:
@CrippleCat can we close this and if you can reproduce please comment and we'll re-open ?

closed 2024-10-07 by luzpaz