FreeCAD/FreeCAD #17018: [Problem] External ARCs can not be comfortably replicated

rabirland, 2024-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Possible relation to #10043 but not quiet the same.

Adding an external ARC to the sketch, then drawing a normal ARC over it (using the external's 3 points) results in a "redundant constraint" error. Removing any of the constraints turns the sketch into under-constrained state. Trying to add any more constraint can easily turn into over-constrained state again.

Workarounds exists but are way too tedious and unnecessary complicated.
The issue can be seen here:
https://www.youtube.com/watch?v=HQ-rnT5tcvQ

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * Curves 0.6.48
  * fasteners 0.5.29
  * freecad.gears 1.0.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

Adding #10043 as a feature would solve this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-05:
I think this is expected as you click on the external points too which creates coincident constraints which would lead to an over-constrained sketch. If you place the arc end/start point just on the external arc, does it work?

rabirland, 2024-10-06:
@maxwxyz using only the start / end points creates an underconstrained arc because the radius is not defined. A workaround is to use the start/end points and then apply an equality constraint on the external arc and the sketch's arc, which could be easily automated or if the implementation of the over-constrained checker allows it, ignored.

arcivanov, 2024-10-18:
I would consider this an outright bug. The arc is defined by 3 points: center and start and end points on the curve. If you match all three points, the sketcher reports "redundant constraint" for one of the points on the arc, if that constraint is removed the arc is underconstrained. The equality of the arc to the external geometry is not sufficient, since equality only defines a radius and does not fully constrain the arc. 

The problem example is attached.

[constrain_arc_bug.FCStd.zip](https://github.com/user-attachments/files/17436693/constrain_arc_bug.FCStd.zip)

arcivanov, 2024-10-18:
To emphasize why I consider it a bug - the arc is defined by a minimum of 3 points. Using those three points is "redundant constraint", which is wrong - there is no excess constraint. Removing the "redundancy" does not fully define the arc.

VincidaB, 2025-01-25:
Testing this on commit `38806`, the third coincidence constraint get's removed by `void removeRedundantAutoConstraints()`, making the arc underconstrained.
Not making this issue solved but changing the behavior.

I can also reach the error `Autoconstraints cause conflicting constraints - Please report!` from :
https://github.com/FreeCAD/FreeCAD/blob/29ea277b3ba4f38ce7dd0a51ca0d4e6e03b65cf2/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h#L991-L998

By doing an arc on a rounded rectangle's corner. With the attached file, on the corner away from the origin.
[ConstraintError.FCStd.zip](https://github.com/user-attachments/files/18546534/ConstraintError.FCStd.zip)

luzpaz, 2025-07-21:
@matthiasdanner any chance you can take a look at this and #22587 ?

PaddleStroke, 2025-07-31:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/13645
The example is different but the root issue is the same : 
Arcs have only 5 DOF not 6. When you create 3 coincidence you are taking away 6 DOF. So it's overconstrained.

So it's not an actual bug, even though I agree it feels very much like one. The question is what we an do about it. For there is discussion in the older issue.

This issue should be closed as duplicate. @maxwxyz

closed 2025-07-31 by maxwxyz